FaradayRF/Faraday-Software #6: Add Contributors Guide

kb1lqd, 2016-10-03:
A contributors guide is needed to help contributors understand the standards for pulling back into the master. Although all the code is not currently compliant the high level overview is:

- Python follows PEP8 standards for code formatting
- Ignore binaries (i.e. .pyc files)
- Follows our layering practices with IO at an interface (i.e. Flask) for applications with the UI abstracted to another level.

kb1lqd, 2016-10-03:
http://blog.technicallyliving.com/2013/11/creating-a-gitignore-file-in-windows/

https://github.com/github/gitignore/blob/master/Python.gitignore

kb1lqc, 2017-02-06:
@kb1lqd is this https://github.com/FaradayRF/Faraday-Software/blob/master/CONTRIBUTING.md complete?

kb1lqc, 2017-04-18:
Yeah actually this is complete! I totally missed your tagging of me and randomly came upon this. Closing.

closed 2017-04-18 by kb1lqc