FaradayRF/Faraday-Software #8: Update All Heritage Code To PEP8

kb1lqd, 2016-10-03:
Per standards we need to update all old code to PEP8.

kb1lqc, 2016-10-03:
I agree with this. Better to start now while we are moving code over to the public Git Repository.

kb1lqc, 2016-10-03:
Proxy.py code updated to pass PEP8 python module check in commit 8e03e23feb31c66b8d49557811c0cf52d3cbf381

hdkmike, 2017-02-18:
Are you guys using any specific scripts to check for PEP8 compliance?

kb1lqc, 2017-02-18:
Yeah @hdkmike we've used the pep8 module: https://pypi.python.org/pypi/pep8

kb1lqc, 2017-04-18:
OK I'll chime in here and see what @reillyeon thinks. I know we haven't fully implemented `Flake8` in our automated testing. I remember mention of a discussion on things like line length that need to be had. Let's have that on this ticket as it's one of the last items we need to address before fully implementing automated checking with Travis-CI. Let's keep this ball rolling though not the most important issue to solve so we can take our time.

reillyeon, 2017-04-18:
I'm down to a short list of PEP8 issues that are less fun to fix. The module work is actually a part of that because one of the things flake8 doesn't like is us modifying the sys.path in the middle of importing modules.

kb1lqc, 2017-04-18:
Ok so we should link this #109 then. I'm happy to take the load off you for implementing these and pitch in taking point.

kb1lqc, 2017-11-04:
@kb1lqd I believe this is has been addressed. Sure we don't check for all PEP8 violations but every PR now maintains the validity of the checks we do have on all the code. Would you agree we can close this?

reillyeon, 2017-11-04:
I think the remaining errors could be fixed relatively easily. E262, E265 and E266 all have to do with comment formatting. E402 can be resolved now that we don't muck with the Python module path at the top of files because we switched to the library and console scripts model. E501 is the only obnoxious one because I've yet to find a tool to automatically apply sensible line wrapping to an existing Python codebase.

kb1lqc, 2017-11-04:
Ahh good point @reillyeon as it probably is worth just doing the rest. Trying to triage how best to work on Faraday python code this weekend to massively improve the user experience. Feel free to let me know any of your thoughts. Not sure this one is high up there on that particular list.