FaradayRF/Faraday-Software #23: Telemetry: Odd ADC Behavior When Plotting Data

kb1lqc, 2016-10-18:
As shown below, there's an obviuos "tick" when plotting ADC data. This could be a poorly implemented parser or it could actually be in the Firmware @kb1lqd. If nothing about the ADC code changed since the Balloon flight then it's probably the software. I could also use the balloon software to cross check this firmware.

[image]

Not critical to fix right now (10/17/2016) but we should get onto this when looking at ADC firmware or telemetry.

- [ ] Cross-checked balloon flight (8/2016) telemetry and proxy software

kb1lqd, 2016-10-18:
@kb1lqc I'd check with the as-flown HAB firmware. I don't believe I've changed anything (other than fixing the temperature char sign issue).

kb1lqc, 2016-10-18:
This issue could be Firmware as noted in https://github.com/FaradayRF/Faraday-Firmware/issues/15

_I'll keep this ticket open until the firmware ticket closes._

kb1lqc, 2017-02-06:
Just closed https://github.com/FaradayRF/Faraday-Firmware/issues/15#issuecomment-277561232 which addresses a similar if not the same issue. We will see if it was fixed. No root cause. Need to get better about systematically handling these issues in the future. Just @kb1lqd and I right now traveling 1,000,000 mph!

closed 2017-02-06 by kb1lqc