FaradayRF/Faraday-Software #27: Device Configuration - Parameter Updates

kb1lqd, 2016-11-02:
While updating the device configuration application and respective proxy module I have updated several fields that should be brought back into the main "faradayIO" repository. This is a result of the faraday_config.ini interpretation and upgrades. I also commented out / updated the FaradayIO library deviceconfig.py file.

- Removed specific port 3/4/5 bitmask functions into a single function
- Added GPIO port 5 self variable

_**Make sure these updates are brought back into the FaradayIO module!**_

kb1lqc, 2016-11-02:
Overall we should work to make the proxy configuration itself (not device) a common file. Right now we have several .ini files for proxy.py, telemetry.py, etc. Probably a different issue ticket but worth noting.

kb1lqd, 2016-11-08:
This enhancement was completed in pull request #28

closed 2016-11-08 by kb1lqd