FaradayRF/Faraday-Software #36: Simply Device Configuration Process

kb1lqd, 2016-11-30:
The process as initial release requires the configuration of identical local unit callsign/id in INI files and to update the tutorial script (lines 18/19). This is confusing and error prone. It should be simplified to a single INI file for local device selection.

kb1lqc, 2017-02-05:
I created a simplified method of performing configuration based on your recent configuration work. It's located in the unboxing branch right now.

https://github.com/FaradayRF/Faraday-Software/blob/unboxing/Tutorials/start/configuring-faraday.md

We still need to simplify this entire process but it's managable now for developers. Not clean, but manageable.

kb1lqc, 2017-11-04:
@kb1lqd I believe this as written has been addressed. We configure with faraday.ini now.