FaradayRF/Faraday-Software #37: Update faraday_config.ini From Device Read

kb1lqd, 2016-11-30:
To facilitate easy updating of units and only requiring the changing of intended fields there should be an update made that allows the application to rewrite the INI file "faraday_config.ini" with the units queried current configuration. This file can then be edited manually and programmed back into the Faraday unit.

kb1lqc, 2017-04-18:
Is this really relevant anymore @kb1lqd ? It was nice at the beginning but we now have `[readconfig.py](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/deviceconfiguration/readconfig.py)`. It would be nice to just save it to an INI file but in the end we should likely be figuring out a much better way to configure all of our software anyways.

kb1lqc, 2017-11-04:
ping @kb1lqd ? Not sure it's worth populating an INI file with read data at this time. Probably better to totally revamp our process.