FaradayRF/Faraday-Software #46: APRS Longitude Dropping Leading Zero

kb1lqd, 2016-12-23:
When using the APRS program in Massachussetts it was realized that the APRS server upload was not accepting the station data, packets were received but APRS.fi was showing currupted packets.

Investigation showed good receive data but the APRS conversion prior to upload shows the leading 0 of the longitude getting dropped. I added debug code to print lat/lon and verified.

kb1lqd, 2016-12-23:
Confirmed by forcing a debug GPS longitude of my location with the leading zero, it worked:

[image]

`07122.99` was being truncated to `7122.99`.

kb1lqd, 2016-12-23:
[image]

Valid APRS.fi uploading.

kb1lqd, 2016-12-23:
@kb1lqc do you have any concerns with this solution of the noted bug? I did modify the formatting you were performing but seems like the reasonable solution.

kb1lqd, 2016-12-23:
Seems like my correction didn't catch all situations correctly, the leading 0 on the LAT not liked?

[image]

kb1lqc, 2016-12-24:
Whew is that n coming from after the W? That's causing this now. Either in my code or lower down in your firmware.Try tracing the lat/long as well as lat dir and long dir to make sure that n isn't in there.---- On Fri, 23 Dec 2016 15:42:53 -0800  Brenton Salmi<notifications@github.com> wrote ----Seems like my correction didn't catch all situations correctly, the leading 0 on the LAT not liked?

kb1lqc, 2017-01-02:
Whew is that n coming from after the W? That's causing this now. Either in my code or lower down in your firmware.

 

Try tracing the lat/long as well as lat dir and long dir 

 

---- On Fri, 23 Dec 2016 15:42:53 -0800 Brenton Salmi<notifications@github.com> wrote ----

 

Seems like my correction didn't catch all situations correctly, the leading 0 on the LAT not liked?

 <https://cloud.githubusercontent.com/assets/339144/21463943/76cb112a-c926-11e6-8b5d-8fff3668d5cd.png>

kb1lqc, 2017-02-05:
Yeah so no longer in Vietnam looking at code on my iPhone trying to figure this out. The "n" after the W in the telemetry packet is good. That is the APRS symbol. Need to track this one down. Something is causing the uncompressed location bug.

kb1lqc, 2017-02-17:
Pretty sure I fixed this in #75 . When that one closes let's see if it also gets this issue.

kb1lqc, 2017-02-17:
Per @kb1lqd comments and testing on https://github.com/FaradayRF/Faraday-Software/pull/77 this issue is complete. Closing!

closed 2017-02-17 by kb1lqc