FaradayRF/Faraday-Software #50: First Proxy Data To Unit Fails

kb1lqd, 2017-01-16:
**Issue**
I've noticed that after starting Proxy the first data sent to the unit always fails. Retrying the transmission of the data is successful as is all future transmissions (to my knowledge).

The program first using proxy displays this error:
[image]

I see proxy log:
[image]

**Proper Operation**
Proper operation would not cause error on the first data transmission to the unit. Although I have not debug breakpointed or scope captured on the Faraday device to confirm I have no reason to believe data is being transmitted to the unit on first attempt. It looks as if Proxy can't handle the first lookup of the unit correctly.

kb1lqd, 2017-02-19:
Confirmed bug as run in tutorial with a "fresh" proxy connection:

[image]

https://github.com/FaradayRF/Faraday-Software/tree/Developer_Tutorial_Update_2-18-17/Tutorials/Python_Developer_Tutorials/foundation/Commanding-Local

kb1lqc, 2017-02-28:
This is related to #60

kb1lqc, 2017-11-05:
@kb1lqd Can you double-check that this was fixed?

kb1lqc, 2017-11-05:
Actually I just checked it @kb1lqd and it works without issue. Closing this ticket.

closed 2017-11-05 by kb1lqc