FaradayRF/Faraday-Software #56: Add "GPS_Present" Bit To GPS Bitmask Parsing

kb1lqd, 2017-02-06:
**Overview**
An unused bit in the GPS bitmask of the FLASH parsing should be updated to communicate the presence of a GPS unit on the Faraday unit. This will be used to easily determine if a GPS should be onboard or not but does not indicate the status of the GPS unit if installed, for that the presence of the telemetry "GPS FIX" is used.

**Implementation Details**
* Update software to parse "GPS_Present" bit from GPS bitmask
* Update device configuration INI file to include GPS_Present bit configuration
* Update firmware for default unit factory reset value

**Operation**
This bit is used simple to query the presence of a GPS.
* GPS_Present = HIGH
 * GPS is present and Faraday should attempt to parse valid GPS string
* GPS_Present = LOW
 * GPS is a not present and Faraday should default to internally saved default location

kb1lqd, 2017-02-06:
Post-pone since simply using default location whenever GPS fix not acquired or no GPS string is likely OK.

kb1lqd, 2017-02-06:
False, need a flash bit since no GPS means no NMEA uart data from unit and thus we'll be stuck without a trigger to trigger the grab from flash or GPS dynamically. Need to add flash bit in and assume it is set correctly.

kb1lqd, 2017-02-06:
Adding as BIT 1 in GPS Bitmask in flash memory:

[image]

kb1lqd, 2017-02-06:
GPS Bitmask default in the factory reset can stay 0x01 since default should be no GPS installed and unit configured to have a GPS.

[image]

kb1lqd, 2017-02-06:
Confirmed addition of bit and operation:

GPS_Present bit HIGH:
[image]

GPS_Present bit LOW:
Lon/Lat set to "123456..." along with defaults for all other fields.

[image]

kb1lqc, 2017-04-18:
Is this implemented? Our manual method of telling a unit if it has a GPS installed seems to work fine right now during Faraday programming.

el-iso, 2017-04-18:
@kb1lqc  Setting the GPS_Present bit worked for me when I was doing setup.

kb1lqc, 2017-11-05:
Hey @kb1lqd lets chat about this one, I can't tell if it's actually implemented haha...