FaradayRF/Faraday-Software #63: Unit reports invalid preset GPS altitude

jbemenheiser, 2017-02-10:
When a unit without a GPS is configured with set coordinates, the GPS Altitude that is returned is invalid unless it has 4 decimal points.

Configuration:
DEFAULT_ALTITUDE = 185
Returns:
"GPSALTITUDE": "185\u0000\u0000\u0000\u0000\u0000", 

Configuration:
DEFAULT_ALTITUDE = 185.000
Returns:
"GPSALTITUDE": "185.000\u0000",

Configuration:
DEFAULT_ALTITUDE = 185.0000
Returns:
"GPSALTITUDE": 185.0,

jbemenheiser, 2017-02-10:
It appears that more than four decimals (maybe 8 total characters) overflows into other GPS data as well
Configuration:
DEFAULT_ALTITUDE = 185.00000
Returns:
"GPSALTITUDE": "\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",

kb1lqd, 2017-02-10:
With data provided looks like an software input sanitation given expected memory format. Will reproduce and then patch.

Thanks!

kb1lqd, 2017-02-11:
# Nominal Configuration

All fields manually set with ideal length data.

[image]

[image]

Verifying presence of bug on all GPS fields:

Configured every default GPS flash location to an invalid (small/large) length.

[image]

[image]

All data looks like "0x00's":

[image]

kb1lqd, 2017-02-11:
In `deviceconfig.py` we do check for maximum length:

[image]

kb1lqd, 2017-02-11:
When programming with less than the maximum/ideal bytes in numeric fields the data sent to the unit looks blank "0's":

[image]

Programming the correct number of bytes in numeric locations shows the intended data:

[image]

kb1lqd, 2017-02-11:
Changing just lattitude from `012345678` to `98765` shows trailing 0's in field as expected.

[image]

kb1lqd, 2017-02-11:
# Paths Forward

The paths forward I see right now:
* Software checking forces exact lengths and errors if not
  * Easy but erroring gracefully will be difficult and also annoying
* Software checking forces exact length and corrects with padding if not
  * Easy and automatic so it lesses annoyance to user
* Firmware forces exact lengths and errors if not
  * More difficult and will use limited firmware space/processing

kb1lqd, 2017-02-11:
Adding print debug statements in deviceconfig.py to show variables that pass max length test but (lat) shorter than max displayed as expected:

[image]

[image]

The data is then passed to a struct packing routine. I think the easy key here is either updating the struct packing routine or pre-packing the data.

kb1lqd, 2017-02-12:
Completed initial bug effort: 14c90dfb1612e116b51e67c5abdfe5eea15970f8

I updated:
* Added length test status and check for global failure to deviceconfig.py so that data too long errors (could update to truncation later if wanted)
* Added other generic length checks for good practice
* Added custom padding packet creation function in commandmodule.py to pad data with specific bytes
* Force all Lat, Lon, Alt to floats and then convert back to string (all NMEA strings contain decimals)
* Padding function on Lat, Lon, Alt add 0x30 (ASCII 0) to each unused byte after data. This matches intended GPS NMEA DMM format in ASCII.

Added information in comments for clarity:
[image]

Verified nominal length and format working:

[image]

Verified that any GPS lat, lon, alt, dir... that is too long will provide debug error and result in NOT programming the unit. (Could change this to truncation later...)

[image]

(Note the "GPS string too long"

[image]

Verified that all fields too short (except DIR's) are padded with ASCII 0's (0x30):

[image]

[image]

**Conclusion**

Bug effort is done unless additional updates are warranted.

kb1lqd, 2017-02-12:
Generic configuration version: ad95cdeb4084d552a1ee966fe94906bbee06b381

kb1lqd, 2017-02-12:
Pull request #68

PR #76 (2017-02-18): GPS Bug 63 + Developer Tutorial Clean
Combining Pull request #68 and a developer tutorial cleanup. Accidentally edited both into a single branch and it's easier to just push them both through.

# GPS Bug

Fixed #63 and updated other similar items.

* Added length test status and check for global failure to deviceconfig.py so that data too long errors (could update to truncation later if wanted) and displays on deviceconfiguration.py script command prompt 
* Added other generic length checks for good practice to key fields in device configuration
* Added custom leading/trailing padding packet c [... cut]
closed 2017-02-18 by kb1lqd