FaradayRF/Faraday-Software #67: Proxy Not Forcing Uppercase Callsign

kb1lqd, 2017-02-11:
When using proxy and the telemetry application I noticed that when I use proxy with lower case callsign that Telemetry cannot find the unit even though it is connected and proxy running. All uses of callsigns should be forced UPPERCASE but proxy is not:

# Configuration 1

Proxy: `kb1lqd-1`
Telemetry: `KB1LQD-1`

Results in ERROR

**Proxy - Configured "kb1lqd-1"**

[image]

**Telemetry - Configured "KB1LQD-1"**

[image]

**Running Telemetry**

Telemetry looks for `KB1LQD-1` from proxy but proxy did not force uppercase for callsign and therefor not found.

[image]

# Configuration 2

Changing Telemetry to lowercase: `kb1lqd-1` results in same error as telemetry properly forces the lowercase to uppercase:

[image]

hdkmike, 2017-02-18:
Would this be easily mitigated by using `.upper()` anywhere you read in a callsign and parse it from the `.ini` and always assume it's is uppercase throughout the software?

kb1lqc, 2017-02-18:
Yeah @hdkmike I started using `.upper()` but we haven't scrubbed for it yet.

kb1lqd, 2017-02-18:
Yes @hdkmike , most/all our other programs convert everything to uppercase (using .upper()) except Proxy. This is a legacy remaining bug to squash!

b00klegger, 2017-03-05:
@kb1lqd 
Was this resolved in [53c9849](https://github.com/FaradayRF/Faraday-Software/commit/53c9849c29b9ab8ac558df21cf71f7fee6cf69f5) ?

kb1lqc, 2017-03-05:
@b00klegger it possibly was. Good catch. One of us should make sure that the Proxy, Telemetry, APRS, and Device Configuration python files convert to upper() when importing callsign & never convert a None type to upper().

kb1lqc, 2017-03-05:
Yes @b00klegger is correct, this issue has been solved. I just testing `Proxy`, `Telemetry`, `APRS`, and `DeviceConfiguration` with lowercase and uppercase callsigns for proxy settings. I do not know if `DeviceConfiguration` properly checks for callsign to be programmed to Faraday as uppercase but testing seems to show it does. That is out of the scope of this issue however.

##Proxy
https://github.com/FaradayRF/Faraday-Software/blob/master/proxy/proxy.py#L336

##Telemetry
https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/Telemetry/telemetry.py#L70

##APRS
https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/APRS/aprs.py#L823

##Device Configuration
I cannot find the exact line where callsign from `deviceconfiguration.ini` is converted to uppercase. @kb1lqd you wrote this application. Where is it?

Once Brent answers we can close this issue.

kb1lqd, 2017-03-06:
@kb1lqc  this is performed in deviceconfig.py in FaradayIO https://github.com/FaradayRF/Faraday-Software/blob/master/Faraday_Proxy_Tools/FaradayIO/deviceconfig.py#L105

This was made a long time ago and the thought was to ensure this was done as close to the programming functions as possible.

kb1lqc, 2017-03-06:
OK not necessarily where I thought it would be but we know it's there. Closing.

closed 2017-03-06 by kb1lqc