FaradayRF/Faraday-Software #89: Update/Re-Format Developer Tutorials

kb1lqd, 2017-02-19:
An effort should be made to update the [developer tutorials](https://github.com/FaradayRF/Faraday-Software/tree/Developer_Tutorial_Update_2-18-17/Tutorials/Python_Developer_Tutorials/1-Basic_Proxy_Interactions_And_Programming) to:

* Verify all still work as intended
* Clean and de-clutter
* Update readme.md tutorials to better fit style of basic tutorials @kb1lqc wrote

## Tutorial Scope / Descope

I'd like to move or remove 
* Tutorial 3 - Device configuration
  * This dives deeping inter the device configuration but is largely convered by the alreaday introduced topic and newly created simpleconfig.py https://github.com/FaradayRF/Faraday-Software/tree/master/Applications/deviceconfiguration
* Tutorial 6 - Device Factory Reset
  * I think we should just move this to a special tutorial section of odd's and end how-to's...

[image]

@kb1lqc what do you think?

kb1lqd, 2017-02-19:
I just verified that each tutorial does work as intended (due to massive project updates in the past couple months). I moved the folder structure around and delete unneeded tutorials.

My next steps are to reformat both the readme.md tutorials and update to better code formatting/implementation. I believe that I want to update every tutorial example to use a common INI file structure to match our applications.

kb1lqc, 2017-02-19:
@kb1lqd I agree this seems like a valid path forward.

- Remove everything that the quickstart guide in Tutorials/start covers. You can always link back to those documents where needed in your developer guides.
- Packet formats need basic documentation i.e. simply document UART packet structure on firmware github repository, maybe software too. Documenting doesn't require immediate tutorials on how to use it right now either. We've abstracted a bunch so we can teach applications with the software without needing to understand the UART format for example.

I think we should also consider the topics of the developer tutorials. We have several very distinct paths a developer can go on. Two different developers maybe not be interested in the same path or required to know everything about another path.

Developing to achieve...
- RF connection and data transfer in mind.
- User interface applications.
- Data collection and querying
- Interfacing other services (i.e. APRS application doesn't need developer to understand file transfer)

Are there distinctly relevant paths we should develop tutorials for instead of forcing anyone who wants to develop on the GUI to go through the packet handling tutorials or risk missing out on a basic concept?

kb1lqd, 2017-02-19:
OK @kb1lqc thanks! I've started cleaning and updating, I just finished https://github.com/FaradayRF/Faraday-Software/tree/Developer_Tutorial_Update_2-18-17/Tutorials/Python_Developer_Tutorials/foundation/Commanding-Local

As for the path:

* I'm going to implement basic formatting and cleaning only
* Implementing the INI file configurations

I'm going to postpone for another issue enhancement push for adding new tutorials:

* Finishing ARQ based messenger * Added new paths
  * Yes, I think at this point the basics are covered and give a good idea how to work with Faraday. Actually, the final ARQ messenger is mostly the FLASK version of the simpler prior tutorial with ARQ added as well. I think as this point we can branch into distinct paths (or people can jump to where ever)

I want to capture a good update in this push but I don't want to bog it down too long as it has lost of good working examples. I'd rather give individual pushes for each addition, lets focus on formatting and such here.