FaradayRF/Faraday-Software #90: Add RF Telemetry Tutorial

kb1lqd, 2017-02-19:
Add a tutorial section/page over-viewing how to config, setup, and use a Faraday digital radio with the telemetry application and RF transmissions. This should be really as simple as explaining how to configure the RF beacon enable, power, frequency, and interval timing in the device configuration. It may be wise to also give general tips and practices.

This should probably be written by @kb1lqc  to maintain the flow of wording in the /start, telemetry, and aprs tutorials.

kb1lqc, 2017-02-25:
Starting to work on this. My path forward is as follows:

- Place this guide in the [QuickStart](https://github.com/FaradayRF/Faraday-Software/tree/master/Tutorials/start) tutorials, after the telemetry viewing and LED commanding guides
- I'll largely follow [configuring-faraday.md](https://github.com/FaradayRF/Faraday-Software/blob/master/Tutorials/start/configuring-faraday.md) except it will be written to configure the RF remote unit
- Instructions will include both GPS enabled and non-GPS enabled radio configuration
- I will follow #110 `xxxxxx.sample.ini` convention for configuration

kb1lqc, 2017-02-27:
Completed with #114

closed 2017-02-27 by kb1lqc