FaradayRF/Faraday-Software #91: Telemetry .upper() in Stations Causing APRS to not start

kb1lqd, 2017-02-19:
I noticed that APRS will not start with the latest telemetry.py update. The APRS window crashes and programs fails to start.

[image]

Bug tracked to the "getStations()" telemetry call that APRS tries to use.

[image]

Telemetry.py prompt reports this failure: 
[image]

I printed the callsign variable at line 357 and it was `NoneType`

I believe this is failing because the `.upper()` assumes a string but got `None`. This was recently merged and this situation (using APRS) probably wasn't vetted properly.

**Actions**

I'm going to push a simple pull request to remove the .upper() to allow the program to work. This may cause failure if the user doesn't enter their callsign in all UPPERCASE in the configuration(s).

kb1lqd, 2017-02-19:
Removing .upper() allows startup but no APRS upload yet. Waiting for GPS lock on this unit:

[image]

I'm going to try to verify if I can solve this quickly without removing .upper() since it seems the getStations() isn't getting callsign first. The program is still failing non-gracefully.

kb1lqc, 2017-02-19:
Not sure if actually caused by upper() as you should a I'll be getting a callsign. Is something causing the callsign not to actually be in the dictionary?

kb1lqd, 2017-02-19:
@kb1lqc 

Well I think a different issue is that Telemetry/APRS fails hard if it doesn't see data. It should fail gracefully/not stop.

Tracking this down in aprs.py:

[image]

[image]

Leads me to telemetry.py:

Looks like default if no URL argument supplied is `None`?

[image]

Immediately below this is the failing code:

[image]

kb1lqd, 2017-02-19:
Hmmm... I placed the .upper() back in and ran telemetry (this time with a working and valid GPS signal) and the program doesn't quit but does not log into the APRS servers to upload:

[image]

kb1lqd, 2017-02-19:
Note: I have not yet been able to update APRS location, I was 1-2 days ago.

Interesting:

**APRS Configuration ROTATE.APRS.NET**

Although it failed originally to cause this ticket in this default APRS server it no longer exhibits the .upper() failure noted when configured to use this server.

**APRS Configuration SECOND.APRS.NET**
When using this server (prior commits) and correct port the .upper() bug reappears.
[image]

kb1lqd, 2017-02-20:
So APRS is failing at line 93 which is the:

`results = r.json()`

[image]

Notice that `url = "http://" + host + ":" + port + "/stations"` forms the GET request URL without any arguments? This is probably why they are being passed as "None." Although telemetry ( @kb1lqc ) telemetry should handle/fail gracefully in this situation we should avoid APRS from causing this condition in the first place.

kb1lqd, 2017-02-20:
OK I think I have an idea of whats happening, I moved the telemetry.py parameters dictionary filling code into the try/except bock to fail softly.

[image]

I also printed debug information.

[image]

Clearly everytime the `/stations` is accessed the following parameters are `None`
* startTime`
* `endTime`
* `Callsign` 

The try/except however cause the script to not fail and somewhere the stations are found and APRS is **actively working now!**

kb1lqd, 2017-02-20:
Debugging further it is clear that parameters contains type "Nones" as expected entering the `queryStationsDb`

Clearly `queryStationsDb` is providing the data from the database which is why this is now working. Why is this needed? What is the URL callsign/parameters for @kb1lqc ?

[image]

[image]

kb1lqd, 2017-02-20:
@kb1lqc I went back to the master and still tried but getting connection errors with and without .upper(). I need you to review and fix Telemetry and APRS, they seem to be broken from an APRS side.

kb1lqc, 2017-02-20:
Hmm so here's the gist of APRS:
- stations/ returns a list of any callsign-nodeid that have been heard in the last 5 minutes
- querystationsdb() takes in a list of stations and obtains the latest detailed telemetry from each station.

I'll have to look at this on Monday and see what's going on. This worked fine for me when I tested it. Are you sure you're fully configured?

kb1lqc, 2017-02-20:
OK I am looking into this now @kb1lqd.

h1. telemetry.py

It appears that line 124 hows the root telemetry server response "/" as defaulting to "%" for callsign. This is an SQLite wildcard. Basically "%" for callsign will return anything while "KB1LQ%" will return any database SOURCECALLSIGN that begins with "KB1LQ[a-z,0-9]"

[image]

This does not appear on the "/stations" app route:
[image]

h1. Investigating
I placed a `print callsign` statement on line 337 of telemetry.py and a `print r.text` on line 89 of aprs.py. Below is my standard and properly configured APRS application on the bottom and the top command prompt being Telemetry.

[image]

It's clear that the `None` type in Telemetry from doesn't seem to matter at least on my machine. It works fine!

Printing out the SQL query when a "None" callsign is presented on [line 607](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/Telemetry/telemetry.py#L607).

[image]

It doesn't matter if "None" is the callsign, at least once the upper() is removed since that would have failed.

kb1lqc, 2017-02-20:
BTW @kb1lqd when your APRS application did this:
[image]

That's indicative of you not connecting to APRS-IS. So that problem had nothing to do with telemetry. At least in that instance.

kb1lqc, 2017-02-20:
OK interesting I noticed that if I type the wrong APRS passcode into login.ini (my personal APRS-IS login info for our APRS application) the connection will still proceed nominally

[image]

However, no data ever makes it to the APRS-IS server...

[image]

kb1lqc, 2017-02-21:
OK @kb1lqd I pushed updates to my [issue91 branch](https://github.com/FaradayRF/Faraday-Software/commit/1ff5e1a4b4d2999ec1a781df41cdcdf6d4d63cc2) which performs the following.
- Cleaned up telemetry.py upper() statements
- Forced upper() on telemetry callsign URL paramerter get() calls
- Changed callsign default value from "None" to "%" which passes an upper() call and is a valid SQL wildcard
- Removed SQL string generation for callsign = None
- No longer uses login.ini configuration file for APRS, APRS-IS callsign located in aprs.ini now
- Implemented a basic APRS-IS passcode generator. This automatically generates a passcode to protect against simple poor passcode entry and helps remove an extra configuration file.

kb1lqc, 2017-02-21:
Interesting, as I let the APRS application sit there it eventually stopped. It appears as though APRS-IS forcibly closed my connection. I wonder if it's because of the update rate?

[image]

kb1lqc, 2017-02-21:
Yeah I now fail gracefully but I'll have to see if it's because of the 30 second rate I performed this test on...

[image]

kb1lqc, 2017-02-21:
OK I changed the time interval to 60 seconds and still received an ERRNO 10054:

[image]

At this point it's probably a case where a simply catch of reconnecting to APRS-IS when this error occurs would fix it. However, @kb1lqd what do you feel about Issue #96 where we could connect to APRS-IS and perform APRS functionality using the aprslib.

Honestly I don't care enough about providing a fully implemented APRS suite. Faraday is my main goal and if someone else was passionate about implementing a stable APRS library then I feel we should leverage their work, credit where credit is due, and move on!

hdkmike, 2017-02-22:
One thing that you can do is:
```
PARAMETERS["CALLSIGN"] = callsign.upper() if callsign else None
```
This essentially only evaluates `callsign.upper()` if the condition (`callsign`) is met. If it's not, it just evaluates to `None`.

kb1lqd, 2017-02-22:
Mike,

 

Yes, GPS isn’t locked and 0’s are being sent from GPS. We haven’t implemented a default to default location when GPS not present. That could be bad if just randomly switching and better left IMHO sending zero’s. You can set the “GPS Present” bit to 0 and the GPS location will default to the flash config.

 

For RF Telemetry:

 

We were questioned on this already so here’s a quick how to:

I realized that Bryce never put this in his tutorial and I created a planned update to do it: https://github.com/FaradayRF/Faraday-Software/issues/90

 

The Faraday units as shipped are not configured to automatically transmit over RF (no valid Callsign). It’s easy though to get RF telemetry working through device configuration.

 

Following the device configuration tutorial <https://github.com/FaradayRF/Faraday-Software/blob/master/Tutorials/start/configuring-faraday.md>  of the unit you want to transmit RF telemetry beacons (both can simultaneously if desired). Make sure to set the RF telemetry bit to “1” (this is default 0). You can also update the telemetry transmission interval (in seconds). Also, RF power output is relatively low by default (due to possible desense if sitting close on a table), if you like to crank it up the maximum “BOOT_RF_POWER” is 152. Manually power cycle the device to have effect. 

 

 

Expect a more detailed tutorial in the next day or two. Also, that developer tutorial branch I linked might be of interest to jump into the programming side.

 

*	Brent

kb1lqd, 2017-02-23:
Pull request #93 looks to have fixed the issue! I no longer get errors or crashing program.

closed 2017-02-23 by kb1lqd