FaradayRF/Faraday-Software #96: Consider aprslib for APRS Application

kb1lqc, 2017-02-21:
I just became aware of [aprslib](https://aprs-python.readthedocs.io/en/latest/) which looks like it takes care of a lot of the APRS functions. Also, it's open source and [on Github](https://github.com/rossengeorgiev/aprs-python).

We should see if it implements APRS-IS connection as well as packet forming and parsing better than I did. Trying my best seemed to get me usable functionality but I did not implement much of APRS at all and it would be great to leverage a project that actually focuses on implementing APRS.

This would be a very significant amount of work for the APRS application. However, better now than before we attempt to implement more APRS functionality or squash any lingering bugs that this library inherently solves.

kb1lqc, 2017-02-21:
@rossengeorgiev would you have any input? Am I understanding the utility of your project correctly? Looks like some great work!

rossengeorgiev, 2017-02-21:
Hi @kb1lqc,
I'm glad you like `aprslib`. I made it a while ago for http://tracker.habhub.org/ 
The focus was on APRS-IS and parsing position reports. It been in use at habhub for a couple of years running non-stop looking for balloons, so for that purpose it has been solid and stable. It's pure python and runs on all versions. It deals with most of the craziness of APRS. However, it is by no means complete. There is only some groundwork for generating packets. It doesn't deal with KISS. Doesn't parse all report types. Needs more unit tests. I have no plans to work on it in the near future, but I'm happy to accept contributions.

kb1lqc, 2017-02-21:
Thanks for the prompt response @rossengeorgiev. I've totally seen habhub.org before, very cool. Yes we wouldn't expect anything other than what you currently have or are willing to put into your own project. I'd certainly consider contributing.

## FaradayRF Needs
Our [Proxy](https://github.com/FaradayRF/Faraday-Software/tree/master/proxy) Application takes in data from a [Faraday Radio](https://faradayrf.com/faraday/) which operates on the 33cm ham band here in the United States. The [APRS](https://github.com/FaradayRF/Faraday-Software/tree/master/Applications/APRS) application I wrote queries for telemetry from Proxy then packages it up in an APRS format string (Position, Telemetry, Equations, Labels) then sends the string to APRS-IS. Faraday appears as if it's operating on the APRS network but it's really not, thanks to APRS-IS.

[We've flown Faraday with a balloon over California using this application](https://faradayrf.com/simple-faraday-balloon-launch/)!

I never implemented packet decoding because that wasn't an immediate goal. With time I was hoping to fulfill many of the APRS functions but for now kept it extremely barebones.

## `aprslib` & Faraday
It appears as though you largely approached APRS from the other end, decoding packets for display on your website, not sending them into APRS-IS.

Immediate useful to us include:
- [Sending APRS-IS Packets](https://aprs-python.readthedocs.io/en/latest/examples.html#sending-a-packet)

It appears you provide no means to actually generate an APRS string, is this correct?

I actually already implemented an [APRS position packet formatter](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/APRS/aprs.py#L164) among others.

If we decided to integrate `aprslib`, especially with the future goals of decoding packets too, I see us possibly adding functionality to send packets into APRS-IS to your project. Would this be something you'd be willing to let us suggest?

rossengeorgiev, 2017-02-21:
You can generate simple position report, but that could easily be extended for other formats.

https://github.com/rossengeorgiev/aprs-python/blob/f6a952a1091963dc7bbe4721e83b00f8df02d44e/aprslib/packets/position.py

I've taken a more generic approach. Fill in the bits you want, and the class will serialize itself. You can just pass the instance to `sendall`. Contributions that extended are welcome.

kb1lqd, 2017-02-22:
@kb1lqc  Can we try a quick update to aprslib to perform basic functionality? To my understanding the only part of the "APRS" script we'd be replacing is the actual hooking into the APRS-IS system but we'd be keep our RESTful interface code?

kb1lqc, 2017-02-22:
@kb1lqd I can put it on my todo list. I believe there's basic position string generation available with [position.py](https://github.com/rossengeorgiev/aprs-python/blob/f6a952a1091963dc7bbe4721e83b00f8df02d44e/aprslib/packets/position.py) as pointed out.

Unless I'm completely misreading the code that appears to generate a string at the end with the appropriate information for an APRS position report.

rossengeorgiev, 2017-02-22:
Basically works this way

```python
In [2]: p = aprslib.packets.PositionReport()
In [3]: str(p)
Out[3]: 'N0CALL>N0CALL:!0000.00N/00000.00El'
In [4]: p.fromcall = 'LZ1DEV'
In [5]: p.tocall = 'LZ1DEV-3'
In [6]: p.latitude = -4
In [7]: p.longitude = 23.5
In [8]: p.comment = "test comment"
In [9]: str(p)
Out[9]: 'LZ1DEV>LZ1DEV-3:!0400.00S/02330.00Eltest comment'

kb1lqc, 2017-02-22:
Awesome!---- On Wed, 22 Feb 2017 10:38:56 -0800  Rossen Georgiev<notifications@github.com> wrote ----Basically works this way In [2]: p = aprslib.packets.PositionReport() In [3]: str(p) Out[3]: 'N0CALL>N0CALL:!0000.00N/00000.00El' In [4]: p.fromcall = 'LZ1DEV' In [5]: p.tocall = 'LZ1DEV-3' In [6]: p.latitude = -4 In [7]: p.longitude = 23.5 In [8]: p.comment = "test comment" In [9]: str(p) Out[9]: 'LZ1DEV>LZ1DEV-3:!0400.00S/02330.00Eltest comment'

kb1lqc, 2017-03-05:
Starting work in [aprslib branch](https://github.com/kb1lqc/Faraday-Software/tree/aprslib) to see if aprslib can be brought in successfully.

kb1lqc, 2017-03-05:
Just create the following issue on `aprslib` to get an answer about the intended function of coordinate conversions:
https://github.com/rossengeorgiev/aprs-python/issues/25

kb1lqc, 2017-03-07:
OK I converted telemetry data to Decimal format to work well with `aprslib`. Now I can't seem to get `aprslib` formed packets to upload to APRS-IS....

Looking into this. The working APRS application position report getting into aprs.fi shows the following

`2017-03-06 23:19:47 CST: KB1LQC>GPSFDY,qAS,KB1LQC:!3403.13N\11814.62W#.../000/A=000010Faraday Access Point`

The position string sent to the APRS-IS socket is:

`KB1LQC-0>GPSFDY:!3403.13N\11814.62W#.../000/A=000010Faraday Access Point`

That is interesting that the 0 was stripped

kb1lqc, 2017-03-07:
Oh this is dumb... APRS-IS or APRS.fi is stripping -0 callsigns down to just bare callsigns. The SAME code with Faraday programed as `KB1LQC-1` Results in 

APRS.fi RAW
`2017-03-06 23:27:00 CST: KB1LQC-1>GPSFDY,qAS,KB1LQC:!3403.13N\11814.62W#.../000/A=000020Faraday Access Point`

Faraday APRS Position String:
`KB1LQC-1>GPSFDY:!3403.13N\11814.62W#.../000/A=000020Faraday Access Point`

OK I'll test the new code with this configuration.

kb1lqc, 2017-03-07:
WHOA! This only works when I use `aprslib` socket connection and `sendall()`... why? Valid data got into APRS-IS:
`2017-03-06 23:54:23 CST: KB1LQC-1>GPSFDY,qAS,KB1LQC:!3403.13N/11814.62Wn/A=000066Faraday Wireless Node`

rossengeorgiev, 2017-03-07:
> That is interesting that the 0 was stripped

It is redundant. Your callsign without ssid implies `ssid = 0`

kb1lqc, 2017-03-07:
Yeah apparently, I would have expected it to keep the zero since I specified it but I was wrong.

kb1lqc, 2017-03-11:
I've submitted aprs-python [Pull Request #27](https://github.com/rossengeorgiev/aprs-python/pull/27) adding basic telemetry functionality to aprslib.

kb1lqc, 2017-03-11:
So far APRS application uses `aprslib` to connect to APRS-IS, send a position report and send a telemetry report. The telemetry report is bleeding edge = `aprslib` currently doesn't support it until my code is pulled in on the PR mentioned in the previous comment.

Next up is writing the telemetry labels function for `aprslib` and re-writing APRS for it.

kb1lqc, 2017-04-06:
This issue ticket directly affects #119