FaradayRF/Faraday-Software #97: FaradayRF Software Issue Template

kb1lqc, 2017-02-21:
In an effort to start making issues and work a bit more clear we want to have an issue template to help clearly identify issues/bugs/features. [Github supports an issue template](https://help.github.com/articles/creating-an-issue-template-for-your-repository/) which can be visible or hidden.

- I'd like to hide the template, no use in having it visible to most users. This ticket documents it anyways.
- It should clearly identify the issue with a summary followed by a more in-depth explanation
- A simple explanation of what hardware if any was used when the bug was observed
- Understanding what version/branch/commit of the software is also useful

kb1lqc, 2017-02-21:
Hey @kb1lqd I created an initial template here:

https://github.com/FaradayRF/Faraday-Software/blob/issuetemplate/.github/ISSUE_TEMPLATE.md

What do you think?

kb1lqd, 2017-02-22:
@kb1lqc that looks sweet! I didn't realize you could do templates.

I like the general layout, less is more. So if we approve this ticket you'll unhide the template?

hdkmike, 2017-02-22:
Issue templates are awesome!

kb1lqc, 2017-02-22:
@kb1lqd no if we approve this ticket it remains hidden but every time you generate an issue ticket from then on it will be pre-populated with this content as a framework.

kb1lqd, 2017-02-22:
@kb1lqc Right, I think we're saying the same thing. It becomes the "template" for the new IT's.... Sounds good to me.

kb1lqc, 2017-02-22:
@kb1lqd the file remains in `.github` so you wouldn't see that folder in windows explorer if you just opened the software folder. It's going to remain hidden in that regard but it will be active and affect issue tickets. Basically people won't know it's downloaded in their git repository clone.

kb1lqc, 2017-02-22:
Merged and works like a charm. We can update the formatting as we use it.

closed 2017-02-22 by kb1lqc