FaradayRF/Faraday-Software #98: Add User Dependent .INI files to Gitignore

kb1lqc, 2017-02-21:
All our configuration INI files are not included in the [.gitignore](https://github.com/FaradayRF/Faraday-Software/blob/master/.gitignore) file. This is because we want to track the files and they were under heavy development. However, we are now slowing down on these files changing and it is relevant to remove them so we do not accidentally commit them with our own serial ports and callsigns.

When these files are updated we can simply force Git to add them for that single special commit.
`git add --force my/ignore/file.foo`

## Tasks

- [ ] Ensure all current .INI files in [Proxy](https://github.com/FaradayRF/Faraday-Software/blob/master/proxy), [Device Configuration](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/deviceconfiguration), [Telemetry](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/Telemetry), and [APRS](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/APRS) use "REPLACEME" for `CALLSIGN`, `NODEID`, and `COM`
- [ ] Update [.gitignore](https://github.com/FaradayRF/Faraday-Software/blob/master/.gitignore) to ignore the specific INI files in these documents which change often and should be ignored since they change with every user (callsign, nodeid, etc).

kb1lqc, 2017-02-24:
Checking the `.gitignore` addition and the file seems correct in #110 . You correctly implemented the issue fix. However, it does not actually cause Git to ignore the files since we're already tracking them. [Source](https://help.github.com/articles/ignoring-files/).

Actually, look at: https://stackoverflow.com/questions/1329291/ignoring-an-already-checked-in-directorys-contents#1330097

`git update-index --assume-unchanged path/to/file.txt`

This seems to be more along our needs. However I'm not totally sure that it would not require every developer to run this after checking out the project. @hdmike any suggestions?

kb1lqc, 2017-02-24:
I think I could have written a better Issue ticket in hindsight. I don't think we absolutely need the .gitignore additions. The result we want is to have git keep a specific version of the file in the repository but not allow more changes unless a change to that files is specifically called out as being worthy of updating.

Maybe this is how the original idea works with `git add --force` though I'm not convinced right now. Once we determine the path forward on this I think #110 can be pulled in.

jbemenheiser, 2017-02-25:
This isn't a new problem.  There are several recommended solutions here.  (https://gist.github.com/canton7/1423106) 
It seems the two leading options are to make templates or add the git update-index --assume-unchanged [file] as a setup step in the guide and scrub for commits to the configs on every pull request.  
I think that your original approach of tracking proxy-template.ini in git and having each user save it locally as proxy.ini without tracking it makes the most sense.  The proxy.ini file would need to be removed  from the repository and remain on the .gitignore list.  This is easier to maintain in the long run.  The problem with this solution  is when you make changes to the template, users would not necessarily know that they needed to update their local file unless it breaks the program.  However, if the template is stable, it would work well.  
I am interested in working on a solution that works for all on the config files in the project once we pick a direction.

kb1lqc, 2017-02-25:
Thanks for the info @jbemenheiser. You've convinced me that at least for the beginning of this project we should use the template or sample files method that during configuration must me copied into a non-sample or non-template `.ini` file.

Using a `xxxxx.sample.ini` actually sounds good to me. It's shorter and more to the point.

While we probably should address this on another issue ticket and think about it for a bit one way I see dealing with out of date templates is to have a single .ini file in all applications that simply has the latest version number of a configuration file in it while all `xxxxx.sample.ini` files have their own version number embedded. The sample would be read every time just for this value to check if it's different from the user generated configuration file and warn them if it is.

## Path Forward
OK. @jbemenheiser here's what I'd like you to do on #110.
- Create a `xxxxx.sample.ini` file for each application currently used (Proxy, Telemetery, APRS, and Configuration).
- While you're at it I'd appreciate it if you could knock off #112 when you update the sample configuration files thus hitting two birds with one stone.
- `git rm -r --cached <filename>` user-dependent and/or old .ini files we no longer are tracking. Those are actually all the ones listed on #112 with checkboxes
- Ensure `.gitignore` has the non-sample version of the configuration files in them.
- Update documentation for [quickstart](https://github.com/FaradayRF/Faraday-Software/tree/master/Tutorials/start) to move/edit `sample.ini` config files
- Update Application `readme.md` files to move/use `sample.ini` config files
-- APRS [readme.md](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/APRS/readme.md)
-- Telemetry [readme.md](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/Telemetry/readme.md)

jbemenheiser, 2017-02-25:
Working in my issue98 branch, I created the config.sample.ini files scrubbed them per #112 and removed the config.ini files.  

However, I overlooked a major flaw in this plan.  Since the config.ini files are ignored by git, they are removed when you checkout a different branch. The config.ini files would need to be recreated from the config.sample.ini files every time you switch branches.  Not good.

kb1lqc, 2017-02-26:
Good point @jbemenheiser. We need to think about this more... Turning out to be quite the problem to solve. We can't be the first!

jbemenheiser, 2017-02-28:
This issue has been haunting me since Saturday and my comment above because all the online sources I found confirmed that the way I implemented the change should have worked.  I asked around a bit with people who know their way around git better than I do, and they agreed that this should have worked.  So, I played with it a bit more and found the answer. 
 
When I tested the config setup previously, I was working from my commit that had added the config.sample.ini files, removed the config.ini files and added the config.ini files to the .gitingore list.  To test it, I modified the config.sample.ini and saved it as config.ini.    I expected to be able to switch to a different branch and switch back and still have my config.ini file.  However, when I switched back, the config.ini was not there. 
 
This was apparently due to the fact that the branch that I switched to for the test was still tracking the config.ini file.  Since one of the commits I had made was to remove the config.ini and add it to the .gitignore list, that change was pulled in when switching back and the file was removed.  If both branches had included that change, the untracked file would have remained in place while switching.
 
I tested this again by creating the sandbox branch and merging the config file changes into it.  It is now possible to make changes to the config.ini files in the issue98 branch and switch over to the sandbox branch and back without losing my configs.  I also confirmed that when I switch to my master branch that is still tracking the config.ini files and switch back to the issue98 branch, the config.ini files were removed. 

The problem was in the way I tested it not the way I implemented it.  Next, I will start updating all of the guides so that pull request #110 can be reviewed and merged sooner rather than later.

kb1lqc, 2017-03-04:
Fixed in PR #124

closed 2017-03-04 by kb1lqc