FaradayRF/Faraday-Software #99: Cleanup Proxy Folder Name To Use Capitol "P"

kb1lqc, 2017-02-21:
[Proxy](https://github.com/FaradayRF/Faraday-Software/tree/master/proxy) is currently in a folder `proxy` which is not consistent with other applications that use an uppercase letters. I'm open to being convinced we should move to all lowercase letters too however this is what I picked. If you feel strongly please suggest a reason! Basically suggesting that core applications for Faraday should be uppercase folder names.

[image]

## Tasks

- [ ] Rename proxy folder
- [ ] Check that quickstart guide for[ configuring and running Proxy](https://github.com/FaradayRF/Faraday-Software/blob/master/Tutorials/start/configuring-proxy.md) is still valid. (If you don't own a Faraday radio simply ask @kb1lqd  or myself to test it)

kb1lqc, 2017-03-07:
Also, @amrithasuresh the proxy.md file references lower case proxy in it's folder comments too if you would like to help do another quick PR to address the two places were they are referenced.

` https://github.com/Amrithasuresh/Faraday-Software/blob/master/Proxy/readme.md
` https://github.com/Amrithasuresh/Faraday-Software/blob/master/Tutorials/start/configuring-proxy.md

Amrithasuresh, 2017-03-08:
I.

In Faraday-Software/Proxy/readme.md file. Under "**Configuration**" section in "line 19"

1. Navigate to the `proxy` folder

Have changed 'proxy' to Proxy

II.

In Faraday-Software/Tutorials/start/configuring-proxy.md. Under "**Windows**" section in "line 38"

1. Navigate to the `proxy` folder

Have changed 'proxy' to Proxy

III. 

In Faraday-Software/Tutorials/start/configuring-proxy.md. Under "**Linux (Debian-Based)**" section in "line 54"

1. Open `proxy.sample.ini` with a text editor i.e `gedit git/faradayrf/software/proxy/proxy.sample.ini`

Have changed 'proxy' to Proxy

IV.

In Faraday-Software/Tutorials/start/configuring-proxy.md. Under "**Mac OS X**" section in "line 66"

1. Open `proxy.sample.ini` with a text editor `faradayrf/software/proxy/proxy.sample.ini`

Have changed 'proxy' to Proxy

V.

In Faraday-Software/Tutorials/start/configuring-proxy.md. Under "**Connecting Proxy to Faraday**" section in lines 80,83,86

~faradayrf/software/proxy/proxy.py

Have changed 'proxy' to Proxy.

Is that right?. So please point me. If it is correct then I will give a pull request. 

Also, one commit for one file right?. Thank you.

kb1lqc, 2017-03-08:
Yeah this looks good. Thanks for the thorough planning here. For the sections in configuring-proxy.md where proxy is in the path i.e. `faradayrf/software/Proxy/proxy....` I take it you meant to change the proxy that is the folder in that path and not the filename as I did in my example. If that's the case we are in agreement.

Should be two files, the readme.md in the Proxy folder and the configuring-proxy.md in the tutorials/start. You can certainly do this in one commit but I really don't mind how many it takes :). One new PR with these updates would work nicely.

Thanks!

Amrithasuresh, 2017-03-08:
>For the sections in configuring-proxy.md where proxy is in the path i.e. faradayrf/software/Proxy/proxy.... I take it you meant to change the proxy that is the folder in that path and not the filename as I did in my example. If that's the case we are in agreement.

@kb1lqc I did it the same way. Thank you.

kb1lqc, 2017-03-09:
Thanks for the help @Amrithasuresh. Always welcome to keep lending a hand. I hope you one day consider getting your amateur radio (ham) license one day if you don't already. We're trying to make it much more interesting for programmers and digital experimentors. www.faradayrf.com

closed 2017-03-09 by kb1lqc