FaradayRF/Faraday-Software #108: Create a COM Port Testing Program

kb1lqc, 2017-02-24:
## Summary
We should be able to "fake" being a Faraday radio by looking like a COM port sending valid UART data. This would allow us to automate testing and iterate without hardware. It would also allow people without Faraday hardware to use the software and potentially help contribute back to the project.

## Problem Explanation
The following Stack Overflow summarizes the options nicely:
https://stackoverflow.com/questions/1605721/faking-an-rs232-serial-port

### Environment
#### Software 
This should be cross platform but we are largely developing on Windows at this time. I would love to see Windows, Linux, Max OS X support at some point.

#### Hardware
The goal is to remove the need for hardware!

# Supporting Information
N/A

kb1lqd, 2017-03-07:
@kb1lqc  Is this important? Over dev tutorials?

kb1lqc, 2017-03-07:
It is important, @jbemenheiser seemed interested in taking this on slowly over time when I talked with him this weekend. Might make sense to do less of a COM port simulator and more of an addition to Proxy which saves BASE64 strings of at least telemetry (for starters) so it can be served up in a "debug" mode of Proxy that doesn't connect to serial COM and just sends what's in the database.

jbemenheiser, 2017-03-07:
I am interested in working on this issue.  My understanding of the task was to create a mode within Proxy that can send out data for other applications without hardware connected.  

Hardware Simulation Mode:
* Make a database with sample data in BASE64 strings that has changing values
* Within proxy.py, add functions to read in data from the database and send it out at the configured rate
* Within proxy.sample.ini configure simulation mode and datarate

I would probably still start with running through the developer tutorials first.  I don't know the code well, so I need to get up to speed there.  Having this task in mind as I go through the tutorials will give me an external point of reference.

b00klegger, 2017-03-08:
+1 @jbemenheiser
I have (like others I'm sure) finally cobbled together something in proxy which would seem to make Proxy run in a 'normal' way to use as a base for further development. Nothing as sophisticated as what has been proposed here of course. Ping me if you want some help.

kb1lqc, 2017-03-09:
FYI @b00klegger and others you can use the up-vote emoji which is used as an actual up-vote count on GitHub!

[image]

kb1lqc, 2017-03-09:
Also @b00klegger what do you mean you "finally" cobbled something together to run in a normal way? Have you been playing with some new code? Sounds interesting!

b00klegger, 2017-03-09:
Sorry about the +1, don't use Github often and couldn't find it yesterday.

@kb1lqc Perhaps my use of 'finally' is a bit dramatic. I don't have hardware (yet!) and was looking through issues last week, looking to make some contributions here. I got frustrated that I wouldn't be able to quickly iterate through ideas and fixes. I noticed this issue and decided that yeah, I think I can hack something together.

It has been instructive reading through the stack of proxy <--> layer4 <--> layer2 and getting a glimmer of how that all works.

@jbemenheiser What is your slack handle? I can forward you my hacky approach and some notes to give you a hand.

kb1lqc, 2017-03-09:
@b00klegger oh no problem I only used "finally" since you quoted it too :) No worries on being dramatic it was just hard to tell via text what you were trying to convey! I agree that is a bit frustrating and this would help people contribute/try out Faraday without owning hardware.

An additional thought, this would likely be putting us on the direction to start unit testing because hot damn we need to start unit testing someday ;). Fun stuff and thanks for putting the thought into this!

kb1lqc, 2017-03-10:
@b00klegger I talked to @jbemenheiser today and he's going to be busy for a bit and likely unable to start this. Any interest in taking this one on? I'll certainly help where possible.

b00klegger, 2017-03-11:
I'd be happy to take on the task. I know I can get it done. 
I'll send you some questions over Slack to help me get rolling.
=) =)

kb1lqc, 2017-03-13:
@b00klegger lets chat here so it's public if you don't mind. Here are some of my thoughts.

- It's probably easier to modify Proxy to replay a file such as a sqlite database of BASE64 data saved from some Faraday radio performing a specific task. Let's assume Telemetry from itself (Local) and an RF unit (Remote).
- Instead of trying to mimic the COM port which doesn't seem very straightforward why not get 95% of the way there and just have proxy loop around playing some file and tricking all other applications into thinking Proxy is really connected to a piece of hardware?
- Is this the correct way to move towards unit testing? Or are we complicating it by doing this and do you know of a way to perform unit testing on indivitual RESTful applications? Actually, Crap... [Flask has unit testing for RESTful applications](http://flask.pocoo.org/docs/0.10/testing/).

Given the knowledge of the unit testing for flask and RESTful applications maybe this is really two different tasks (or could be one...)
- Simulate Proxy connected to hardware to provide a way to test downstream applications and/or provie a "Demo" mode were people can use Faraday software and develop with it and not actually own hardware.
- Figure out unit testing on Proxy, Telemetry, APRS, etc.

Thoughts?

b00klegger, 2017-03-15:
@kb1lqc 
Ok. Popping my head up here after noodleing this for a few days. 

I like the idea of replaying a 'known, good' stream, such as something from Telemetry. This would be the easiest, quickest way to get something rolling.

I have been mucking around the past few days trying to get a 'virtual COM port' thingie working, winding my way through the stack following the packet. 

> Instead of trying to mimic the COM port which doesn't seem very straightforward why not get 95% of the way there and just have proxy loop around playing some file and tricking all other applications into thinking Proxy is really connected to a piece of hardware?

Agree!! My brain is swiss cheese. \x00\x00 BB123s
On the bright side, however, I have learned about serial port programming over the past few days ;) .
Eventually it might be helpful to have a virtual port as you would want to see the traffic being sent to the (virtual) board. At first blush this approach could work for Telemetry, Message (text messaging app) and APRS also. 

My initial stab at this used a replay of sorts. I subclassed the uart object and, when the `proxy.ini` has `NODEID=0` kicks off this behavior.

Added to `layer_4_service.py`:
```
class faraday_uart_test_object(faraday_uart_object):
    def RxPortHasItem(self,service_number):
        return True

    def GET(self, service_port):
        try:
            if service_port == 0:
                return "This is a test message dingus"
            elif service_port == 5:
                return '\x03\x00aKE0LXD\x05{\x03\x06\x07KE0LXD04N\x06\x07\x06\x15\x06\x13\x01\t\xe0\x073352.4197N11822.6034W32.20000M0.46020.92\x00\x07`\x07\x8d\x06\x1c\x05y\x05E\x05$\x05\x01\x00\x00\x00\x1b\n\xfe\x00\x00\x1c \x00\x00\x00\x00F\x06\x07KE0LXD\x00\x00\x00\x06\x07(\x14\x06\x13\x01\x14\x11'
            else:
                return "Service Port {}".format(service_port)
        except:
            return False
```
I copied the Telemetry (port 5) value from elsewhere in the project. 

Added to `callsign2COM()` in `proxy.py`:
```
num = int(proxyConfig.get('PROXY', 'units'))
# We need at least one valid configuration for a demo mode. 
# range(0,0) is an empty list
units = range(0, num) if num > 0 else range(0, 1)
```
Since my stab at this depends on node=0 I needed to modify this to get just the one valid config out of the function.

I then call my test object in the `main()` loop in `proxy.py` if `NODEID=0`:
```
unitDict[str(values["callsign"] + "-" + values["nodeid"])] =\
    layer_4_service.faraday_uart_test_object(
        str(values["com"]),
        int(values["baudrate"]),
        int(values["timeout"]))
```
Everything works from this point as GET() has been short circuited and returns whatever I want.

As far as I could tell since `proxy.py` was still making a connection I still needed to have some fake serial port somewhere so I cheated and used `socat`. I *think* this behavior could be tweaked out also by changing .... `Layer2ServiceObject` or even further down in `Receiver_Datalink_Device_Class` ??

> Is this the correct way to move towards unit testing?

I can't really comment on this as I rarely write tests for the python work I do :). From my experience it does help to have a good state to test and write tests against to suss out logic errors elsewhere. 

>Given the knowledge of the unit testing for flask and RESTful applications maybe this is really two different tasks (or could be one...)
>Figure out unit testing on Proxy, Telemetry, APRS, etc.

This seems like two tasks to me but could be worked in parallel.

kb1lqc, 2017-03-16:
I give you much credit @b00klegger for stepping through the UART code. That was written about 1-1.5 years ago when we were getting the Rev B boards up and running and decided to revamp the code. It definitely needs some TLC. I would highly suggest steering clear of this code for several reasons:

1. It's bound to be updated at some point as we clean up the low lever Python to current standards
2. While it's totally valid to do the trickery here, it shouldn't be necessary or the most modular/simplistic.

That said, what you proposed seems legit. Essentially create a new Faraday UART object and give it the ability to realize that it's in test mode and insert new data there. You should feel good about figuring this out, that code is not the most straightforward stuff!

## Proxy Implementation
It appears you also agree this is probably best implemented in Proxy. Here's the simplest way I can think of doing this.

### Better UART Object Exceptions

Proxy sets up the uart Faraday object on [line 373](https://github.com/FaradayRF/Faraday-Software/blob/master/Proxy/proxy.py#L373) and does a not so great job at protecting it. This will eventually be updated to fail gracefully but it is what it is right now. I would think that we could check for a configuration bit in `proxy.ini` prior to this line and prevent the unit from even setting up a uart object.
```python
layer_4_service.faraday_uart_object(
                        str(values["com"]),
                        int(values["baudrate"]),
                        int(values["timeout"]))
```

Simply checking for an item in proxy.ini such as:

```
[PROXY]
TESTMODE=1
```

could be used to skip this step and provide a `UnitDict` with a known test value instead of a uart object. We can then pass this along just as we normally do to the `uart_worker` and then check for it so we know not to actually try a GET() on a uart object.

### Better `uart_worker`
Since the previous suggestion passes a normal dictionary of {key:value} station with a value that is some text or value other than a Faraday uart object we can check for it during the [uart_worker](https://github.com/FaradayRF/Faraday-Software/blob/master/Proxy/proxy.py#L44) function.

```python
# Loop through each unit checking for data, if True place into deque
    while(1):
        # Place data into the FIFO coming from UART
        for unit, com in modem.iteritems():
            try:
                for port in range(0, 255):
                    if(com.RxPortHasItem(port)):
                        # Data is available
                        # convert to BASE64 and place in queue
                        item = {}
                        item["data"] = base64.b64encode(com.GET(port))
                        # Use new buffers
                        try:
                            getDicts[unit][port].append(item)
                        except:
                            getDicts[unit][port] = deque([], maxlen=100)
                            getDicts[unit][port].append(item)
```

It would make sense to me to check for this special value in the `com` check of the `modem.interitems()` and instead of performing the `if(com.RxPortHasItem(port)):` we could instead perform a function that opens up some SQLite database table or similar that has a bunch of BASE64 strings of saved data and play those one at a time, incrementing each time.

Do you get where I'm going with this?

## Before Even Tricking Proxy
We need to save BASE64 telemetry from Faraday port 5 and other data from other ports. So a first step could be to have proxy detect that it should go into a "logging mode" from the configuration file and then when the normal UART `if(com.RxPortHasItem(port)):` is called we also can perform an SQLite save to a new database setup just for BASE64 test data. It could be setup with each row consisting of columns:

```
'FARADAY_PORT','DATA'
```
Then we would have knowledge of which message was sent on what port and could simply query the appropriate port when in test mode to place an item into the port queue.

What do you think of this? It seems like nearly the entire thing could live in `proxy.py` without having to resort to messing with the uart layer code. I know this isn't a true COM port substitute but it gets us most of what we actually need and we can always change the name of the issue ticket 😉

b00klegger, 2017-03-20:
I have something to work with for 'tricking' proxy but I cannot commit it to the repo. I will paste here. There is very little 'new' code as I reused some sqlite snips from `telemetry.py`.

```
# Set global data logger mode boolean if present
try:
    DATALOGGER = proxyConfig.getboolean("TESTMODE", "DATALOGGER")
except ConfigParser.NoOptionError:
    DATALOGGER = False
```
Just after proxyConfig.read(). 

```
# Database Functions
def initDB():
    """
    Initialize database, if not present then create it
    Borrowed from telemetry.py
    """

    logger.info("DATALOGGER mode--Initializing database")
    # Obtain configuration filenames
    dbFilename = proxyConfig.get("TESTMODE", "DBFILENAME")
    dbSchema = proxyConfig.get("TESTMODE", "DBSCHEMANAME")

    # Check if database exists
    if os.path.isfile(dbFilename):
        pass
    else:
        # Open database schema SQL file and execute the SQL functions inside
        # after connecting. Close the database when complete.
        with open(dbSchema, 'rt') as f:
            conn = sqlite3.connect(dbFilename)
            cur = conn.cursor()
            schema = f.read()
            cur.executescript(schema)
        conn.close()
    logger.info("DATALOGGER mode--Database init complete")

def sqlInsert(data):
    """
    Takes in a data tuple and inserts into the proxy test data SQLite table
    Tuple is of the form (INT Faraday port, STR Base64 data)
    """

    # Read in name of proxy test data database
    db = proxyConfig.get("TESTMODE", "DBFILENAME")

    telem = [None, data[0], data[1]]

    # Create parameter substitute "?" string for SQL query then create SQL
    numKeys = len(telem)
    paramSubs = "?" * (numKeys)
    paramSubs = ",".join(paramSubs)
    sql = "INSERT INTO PROXY VALUES(" + paramSubs + ")"

    # Connect to database, create SQL query, execute query, and close database
    try:
        conn = sqlite3.connect(db)
        cursor = conn.cursor()

        # Use connection as context manager to rollback automatically if error
        with conn:
            conn.execute(sql, telem)

    except conn.ProgrammingError as e:
        logger.error(e)
        logger.error(telem)
        conn.rollback()
    except ValueError as e:
        logger.error("ValueError: " + str(e))
    except IndexError as e:
        logger.error("IndexError: " + str(e))
    except KeyError as e:
        logger.error("KeyError: " + str(e))
    except sqlite3.OperationalError as e:
        # TODO: cleanup
        logger.error(e)
        initDB()

    # Completed, close database
    conn.close()
```

From `main()`:
```
# Initialize database if in data log mode
if DATALOGGER:
    initDB()
```

From `uart_worker()`:
```
for port in range(0, 255):
    if(com.RxPortHasItem(port)):
        # Data is available
        # convert to BASE64 and place in queue
        item = {}
        item["data"] = base64.b64encode(com.GET(port))
        # Use new buffers
        try:
            getDicts[unit][port].append(item)
            # Conditionally adding item to test mode database
            if DATALOGGER:
                testModeData = (port, item["data"])
                sqlInsert(testModeData)
        except:
            getDicts[unit][port] = deque([], maxlen=100)
            getDicts[unit][port].append(item)
            # Conditionally adding item to test mode database
            if DATALOGGER:
                testModeData = (port, item["data"])
                sqlInsert(testModeData)
```

kb1lqc, 2017-03-21:
@b00klegger OK first off yes currently only members of @FaradayRF/faradayrf can push to the repository. We've contemplated changing this/adding people to @FaradayRF/developers. We've tested this with one other person but haven't had a chance to decide on a path to go down. Currently the [Fork & Pull Request method](https://gist.github.com/Chaser324/ce0505fbed06b947d962) is the way to go for contributors to Faraday-Software. Let me know if you have strong feelings one way or the other. I think this conversation would benefit greatly from having the diff versus the current master!

That said what you posted seems to make sense, though, I'd like to see the diff to be sure. One thing I did notice was that the uart_worker() try/except statement to save to the queue shouldn't use the except clause to place the data into the queue, it will only work if the try statement fails. This should probably just be standard operation of checking if it should save to a database and if so do it and then push the data to the queue and if it shouldn't then ignore the database and push it to the queue.

Thanks for the help so far!

kb1lqc, 2017-03-23:
@b00klegger and @jbemenheiser  I submitted a PR for your branch that allows you to use Proxy without hardware connected. Please note this is actually incorrect code as explained in the PR. It will however fix a few issues when starting up and allow you to see how Proxy should be working. Take it and run!

https://github.com/b00klegger/Faraday-Software/pull/1

Also @b00klegger I highly suggest quickly drawing up a diagram of how this Proxy should work with this datalogging/fake mode before implementing. This way we have a clear idea of what's going on. You can easily copy/paste/screenshot and place into this ticket as part of the discussion.

b00klegger, 2017-03-29:
[image]

The text along the left does a better job of describing the flow than the control chart on the right. As I started to draw the chart out it started to get more complex (in my head) than I had room for. Additionally I realized a few issues with the code I have committed to my fork https://github.com/b00klegger/Faraday-Software/commit/f95484b223bc49b53c76ebcbf53d67b3b0093c53 

A few issues at this point:
- GH identifies some whitespace issues that I did not see locally. Will investigate and fix those.

- When in 'replay' mode there is a lot of activity selecting / inserting into the database. I choose to move this direction as I thought that keeping a bookmark in the faraday_uart_test_object might be more difficult. Open to considering a different approach here.

- This internal 'dummy' serial port trick only works on Linux. May possibly work on OSX (my work laptop is Apple, will try it out there). This is due to some limitations of Python on Windows (lack of a termios library for instance). I was able to install a program called com0com which can create virtual serial port pairs and appeared to work, superficially. If someone has experience with serial port work in Windows I would defer to their experience.

kb1lqc, 2017-03-29:
@b00klegger ok I am totally up for descoping this issue. I'd rather provide a program that appears as a valid proxy than one that appears as a valid COM port. I can change the name of this issue if needed. I outlined this in [my comment](https://github.com/FaradayRF/Faraday-Software/issues/108#issuecomment-286039717). Doing so apparently also mitigates a bunch of platform specific functionality which I would like to avoid.

When starting on this Issue it wasn't immediately clear this would be so platform dependent so with what you helped point out I am strongly against moving down that path right now as we can totally just plug hardware in to test the UART COM port link as needed.

Per my [PR #1](https://github.com/b00klegger/Faraday-Software/pull/1) we already have a good chunk of the base functionality going. Your diagram will loose the right branch with no "dummy" mode.

I think the diagram could be updated a bit to show the progression of code as you step through [main](https://github.com/FaradayRF/Faraday-Software/blob/master/Proxy/proxy.py#L354) and branch off to make the threaded uart_worker and flask app.

This should sufficiently descope the issue to allow you to focus on detecting logging mode, logging to an SQLite db correctly (Read my PR comments above!), and continue on as normal in Proxy. Sound good?

kb1lqc, 2017-03-29:
@b00klegger with my previous comment are you all set to go? What does your schedule look like to implement this?

I apologize on the fractured topic here in this Issue Ticket, we are all learning how to properly run this project and while it's tempting for me to just give this issue a try and solve it I also want you to learn and come up-to-speed with contributing to Faraday. It's a careful balance. I appreciate your patience.

I think we will start needing this by the weekend time frame so if this isn't pushed that much further by the weekend I might give it a nudge this weekend to help with #135 as we try to gain some momentum again. Let me know if you have objections.

b00klegger, 2017-03-30:
@kb1lqc No problem I can get the minor parts finished up before this weekend. I can also get it trimmed up with the minor details trimmed up. I would like to have someone with an actual board try to run the output logging just to verify that something interesting is happening.

I do not mind the flow of things here. While I have been using python for a few years in an admin capacity I need help learning how to fit into a managed software project. To me `HAM + Python = Awesome opportunity`.

I can get it done by this weekend.

b00klegger, 2017-03-31:
@here 
Can I get this commit looked at:
https://github.com/b00klegger/Faraday-Software/commit/13133a90340fdd6edccf8d70a98656a217826416
I have time to iterate over this ticket during the day today and need to know blow by blow what needs to get done to get this ready for this weekend.

kb1lqc, 2017-04-05:
I changed the name of this Issue Ticket to "Add Functionality to Proxy For Logging Data to SQLite3 Database" as that is a more apt name for this ticket. I decided the scope of this ticket should be reduced as we figure out how to properly test Faraday code automatically.

## Proxy Logging to SQLite3
It became clear that to test Telemetry and APRS functionality one of the best methods will be to save real data and push it through to test. This can be as simple as testing for exceptions with known good data. This method of saving data from the COM port is not terribly useful for other applications as we should be able to construct data pragmatically for that.

kb1lqc, 2017-04-10:
Closing, PR #144 pulled this in. We should open a new PR to actually play this data back at some point.

closed 2017-04-10 by kb1lqc