FaradayRF/Faraday-Software #109: Convert Python Faraday_Proxy_Tools/ProxyIO Code to Modules

kb1lqc, 2017-02-24:
## Summary
[Faraday_Proxy_Tools/ProxyIO](https://github.com/FaradayRF/Faraday-Software/tree/master/Faraday_Proxy_Tools/FaradayIO) needs to be ported into a real python module that can be imported. This should be install-able via `pip -e`.

## Problem Explanation
N/A

### Environment
#### Software 
Master needs to be branched and ported

#### Hardware
Faraday Rev D1 (currently released hardware)

# Supporting Information
N/A

kb1lqd, 2017-03-07:
@kb1lqc  Maybe this is a good quick thing to do?

kb1lqc, 2017-03-07:
This is actually a huge project. I attempted to poke at it one night and a few hours in realized it will rewrite the structure of proxyIO files quite a bit. Not necessarily the methods but how the files are structured and how they reference each others classes.

kb1lqc, 2017-03-07:
You're welcome to try and read up on [Python Modules](http://docs.python-guide.org/en/latest/writing/structure/) as this is a very important IT to work on. It will probably take a week in my estimate.

kb1lqc, 2017-04-10:
@reillyeon this is the IT I was referring to that touches upon moving towards modules.