FaradayRF/Faraday-Software #112: Standardize INI Configuration File format

kb1lqc, 2017-02-25:
## Summary
We need to settle on a configuration format for this project with our use of `.ini` files. The INI configuration file format is and informal one so we can make up some of the rules as long as they play nice with the parser reading these files. Updating existing applications such as `Proxy`, `Telemetry`, `APRS`, and `DeviceConfiguration` are the only necessary folders to work in at this time.

## Problem Explanation
The standard I'd like to settle on is:
```
;this is a comment but there are no inline comments
[SECTION]
NAME=value
```
- Uppercase `[SECTION]` and `NAME` fields. Values can be upper or lowercase
- No spaces before or after `=` separating the `NAME` from the `value`
- Comments must be on their own line and cannot be inline comments

This issue ticket does not refer to any logging .ini configuration files which should be dealt with on a separate ticket because they will require some additional programming to change to uppercase.

### Environment
#### Software 
N/A

#### Hardware
N/A

# Supporting Information
Files that currently need to be updated

- [ ] [proxy.ini](https://github.com/FaradayRF/Faraday-Software/blob/master/proxy/proxy.ini)
- [ ] [Telemetry.ini](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/Telemetry/telemetry.ini)
- [ ] [aprs.ini](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/APRS/aprs.ini)
- [ ] [deviceconfiguration.ini](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/deviceconfiguration/deviceconfiguration.ini)
- [ ] [faraday_config.ini](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/deviceconfiguration/faraday_config.ini)

jbemenheiser, 2017-02-27:
I think pull request #117 wraps up this issue, so this ticket can be closed.

kb1lqc, 2017-02-27:
Closing, This should be added to contributions.md or some other method of ensuring future INI files follow the same style.

closed 2017-02-27 by kb1lqc