FaradayRF/Faraday-Software #113: APRS Doesn't Forward RF Stations Reliably to APRS-IS

kb1lqc, 2017-02-26:
## Summary
It appears that the APRS application doesn't actually send out any data to APRS-IS regarding an RF node with much reliability. It also appears that when the RF update rate is set to 3 seconds or so it is more reliable but when the RF interval is 30 seconds only a single packet of several will make it.

## Problem Explanation
With one radio operating on a 9V battery and another operating attached to the computer as a base station it appears that Telemetry reliably receives data. An RF node set to 30 second telemetry transmission rate while operating with a GPS lock and 9V battery about 15 feet away appears to have made it into the database just fine at ~30 second intervals:

[image]

The problem doesn't appear to be RF or Telemetry getting into the database. I suspect one of the following:

- Telemetry `/stations` app route in Flask is returning bad data
- APRS application is not using `/stations` data correctly to obtain latest telemetry packet from `/telemetry`
- APRS application is not using `/stations` data correctly to generate an internal list of stations

### Environment
#### Software 
Master, Running on Windows 8. Unsure if it's present on other environments but it's likely.

#### Hardware
>Let us know about what hardware you are using if applicable. This includes:
>- Faraday radio serial numbers
SN 7 and SN 40, also confirmed with several other users so unlikely just this unit. 
>- Computer operating system
Likely all

# Supporting Information

APRS took a long time to actually send and APRS-IS update between successful updates. The unit KB1LQC-2 was reprogrammed with a 30 second RF interval after 19:14:09 but was on for at least 10 minutes prior to the successful update at 19:37:18. There were roughly 9 times KB1LQC-2 should have been successfully uploaded to APRS-IS

[image]

kb1lqc, 2017-02-26:
Interesting. Using a 30 second transmission and starting with a fresh telemetry database the first APRS transmission resulted in a `/stations` result of just the local node KB1LQC-1. The second APRS update 60 seconds later resulted in just the KB1LQC-2. Following that we see it's just KB1LQC-1 again...

```
C:\Python27\python.exe C:/Users/Bryce/Documents/GitHub/kb1lqc-software/Applications/APRS/aprs.py
2017-02-25 18:15:05,950 - APRS - INFO - Starting Faraday APRS-IS application
2017-02-25 18:15:05,950 - APRS - INFO - Connecting to APRS-IS as: KB1LQC
2017-02-25 18:15:06,191 - APRS - INFO - Connection successful!
2017-02-25 18:15:06,210 - APRS - INFO - [{u'SOURCEID': 1, u'EPOCH': 1488075303.265, u'SOURCECALLSIGN': u'KB1LQC'}]
2017-02-25 18:15:06,220 - APRS - INFO - [[{u'RFSTATE': 0, u'RTCYEAR': 53511, u'RTCSEC': 9, u'GPSLATITUDE': u'3359.1028', u'RTCDAY': 1, u'GPIOSTATE': 192, u'GPSFIX': 0, u'HABTIMER': 0, u'GPSHDOP': 0.0, u'GPSLONGITUDE': u'11828.1690', u'HABTRIGGERTIME': 7200, u'GPSLONGITUDEDIR': u'W', u'ADC6': 0, u'SOURCEID': 1, u'ADC4': 1753, u'ADC5': 1652, u'ADC2': 2192, u'ADC3': 1880, u'ADC0': 2191, u'ADC1': 1890, u'HABTIMERSTATE': 0, u'RTCDOW': 1, u'ADC8': 2856, u'GPSALTITUDEUNITS': u'M', u'KEYID': 11, u'RTCMIN': 10, u'SOURCECALLSIGN': u'KB1LQC', u'RTCMONTH': 1, u'DESTINATIONCALLSIGN': u'KB1LQC', u'DESTINATIONID': 1, u'GPSSPEED': 0.0, u'RTCHOUR': 1, u'EPOCH': 1488075303.265, u'GPSALTITUDE': 100.0, u'GPSLATITUDEDIR': u'N', u'BOARDTEMP': 25, u'HABCUTDOWNSTATE': 0}]]
2017-02-25 18:15:06,220 - APRS - WARNING - KB1LQC-1 No GPS Fix
2017-02-25 18:16:06,230 - APRS - INFO - [{u'SOURCEID': 2, u'EPOCH': 1488075364.494, u'SOURCECALLSIGN': u'KB1LQC'}]
2017-02-25 18:16:06,236 - APRS - INFO - [[{u'RFSTATE': 0, u'RTCYEAR': 57607, u'RTCSEC': 1, u'GPSLATITUDE': u'3359.7000', u'RTCDAY': 26, u'GPIOSTATE': 192, u'GPSFIX': 2, u'HABTIMER': 0, u'GPSHDOP': 0.95, u'GPSLONGITUDE': u'11826.9478', u'HABTRIGGERTIME': 7200, u'GPSLONGITUDEDIR': u'W', u'ADC6': 1257, u'SOURCEID': 2, u'ADC4': 1646, u'ADC5': 1658, u'ADC2': 1987, u'ADC3': 1810, u'ADC0': 2144, u'ADC1': 1841, u'HABTIMERSTATE': 0, u'RTCDOW': 0, u'ADC8': 2853, u'GPSALTITUDEUNITS': u'M', u'KEYID': 33, u'RTCMIN': 16, u'SOURCECALLSIGN': u'KB1LQC', u'RTCMONTH': 2, u'DESTINATIONCALLSIGN': u'KB1LQC', u'DESTINATIONID': 1, u'GPSSPEED': 0.34, u'RTCHOUR': 2, u'EPOCH': 1488075364.494, u'GPSALTITUDE': 5.3, u'GPSLATITUDEDIR': u'N', u'BOARDTEMP': 14, u'HABCUTDOWNSTATE': 0}]]
2017-02-25 18:16:06,236 - APRS - INFO - !3359.70N/11826.95WN
2017-02-25 18:17:06,246 - APRS - INFO - [{u'SOURCEID': 1, u'EPOCH': 1488075426.124, u'SOURCECALLSIGN': u'KB1LQC'}]
2017-02-25 18:17:06,253 - APRS - INFO - [[{u'RFSTATE': 0, u'RTCYEAR': 53511, u'RTCSEC': 9, u'GPSLATITUDE': u'3359.1028', u'RTCDAY': 1, u'GPIOSTATE': 192, u'GPSFIX': 0, u'HABTIMER': 0, u'GPSHDOP': 0.0, u'GPSLONGITUDE': u'11828.1690', u'HABTRIGGERTIME': 7200, u'GPSLONGITUDEDIR': u'W', u'ADC6': 0, u'SOURCEID': 1, u'ADC4': 1785, u'ADC5': 1696, u'ADC2': 2193, u'ADC3': 1892, u'ADC0': 2205, u'ADC1': 1895, u'HABTIMERSTATE': 0, u'RTCDOW': 1, u'ADC8': 2856, u'GPSALTITUDEUNITS': u'M', u'KEYID': 53, u'RTCMIN': 12, u'SOURCECALLSIGN': u'KB1LQC', u'RTCMONTH': 1, u'DESTINATIONCALLSIGN': u'KB1LQC', u'DESTINATIONID': 1, u'GPSSPEED': 0.0, u'RTCHOUR': 1, u'EPOCH': 1488075423.017, u'GPSALTITUDE': 100.0, u'GPSLATITUDEDIR': u'N', u'BOARDTEMP': 27, u'HABCUTDOWNSTATE': 0}]]
2017-02-25 18:17:06,253 - APRS - WARNING - KB1LQC-1 No GPS Fix
```

kb1lqc, 2017-02-26:
Ok this is looking more like a telemetry issue! I realized that using the API to obtain stations with my web browser at http://localhost:8001/stations resulted in just one station being returned instead of the two. I should be seeing any station heard in the last 5 minutes. Instead I'm seeing the latest station. Most of the time this is KB1LQC-1 since that is operating on a 2 second interval while KB1LQC-2 operates on a 30 second interval.

Eventually after many refreshes I obtained this:
[image]

This is odd. I have no idea when the `/stations` stopped reporting all recently heard stations...

kb1lqc, 2017-02-26:
Oh boy now this is interesting... I literally perform SQL "...LIMIT=1" which inherently limits the number of stations to one that could be returned by telemetry to indicate to APRS how many stations to obtain data from and upload to APRS-IS.

December 7, 2016:
https://github.com/FaradayRF/Faraday-Software/blob/5a1ca881c7dd805474b2f8a40df0a3752c8d47ec/telemetry/telemetry.py#L552

This code checks for `callsign != None` and if it isn't then it limits the results to 1 by adding `LIMIT = 1` to the SQL query. This is OK since you are looking for a specific station. However when this fails because `callsign == None` and then the SQL query left without a limit:

`sqlEnd = "GROUP BY SOURCECALLSIGN, SOURCEID ORDER BY EPOCH DESC"`

February 25th, 2017
https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/Telemetry/telemetry.py#L596

Where the heck did the callsign check go?!
https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/Telemetry/telemetry.py#L589

This will ALWAYS limit results to 1. No bueno. Not really worth me tracking down exactly what commit this happened on right now. I'll just fix this.

I do want to point out that This is why issue #91 would also fail with a `None`... we never checked for it and attempted to use callsign value of `None`. In retrospect this is obvious and should have raised a question as to why we suddenly had `None` failing.

kb1lqc, 2017-02-26:
Alright running this for a bit seems to work. I probably haven't run APRS + Telemetry for more than 10 minutes straight yet but it appears to be working much better.

@kb1lqd  or @jbemenheiser could one of you test out my commit to see if it fixes the symptoms of #113 and possibly the other issue Jake was seeing with crashing Telemetry on SQL insertion?

https://github.com/kb1lqc/Faraday-Software/tree/01dbdbe83bd46d558d857259b083f8d09408dee3

jbemenheiser, 2017-02-26:
@kb1lqc  I am running telemetry from your branch.  It only took 20 minutes to get a corrupt packet, but the script logged the error and kept gathering telemetry.

jbemenheiser, 2017-02-26:
Two thoughts for improved logging:
It would be helpful to log the callsign of the rf unit in the info message of aprs.py.  Right now most of us only have one remote unit, but I could see that changing in the future. 

Is now:
```
2017-02-26 14:50:55,470 - APRS - INFO - !3358.46N/11822.71Wn
```
Could be:
```
2017-02-26 14:50:55,470 - APRS - INFO - KB3SVQ-7 3358.46N/11822.71W
```

If a unit is configured to not have a GPS, it would be nice to differentiate that from a unit that has a GPS but does not have a GPS lock.  I think this would be possible since we program that into the unit, but it might not be reported in the telemetry yet.  (This might be a stand alone improvement to implement later)

Is now: 
```
2017-02-26 14:50:55,470 - APRS - WARNING - KB3SVQ-1 No GPS Fix
```
Could be:
```
2017-02-26 14:50:55,470 - APRS - WARNING - KB3SVQ-1 No GPS Installed
```

kb1lqd, 2017-03-01:
Pull request #118 addressed and fixed this issue.

closed 2017-03-01 by kb1lqd