FaradayRF/Faraday-Software #130: Create Basic Text Messaging Program

kb1lqd, 2017-03-14:
## Summary
Utilizing Python only create a basic text messaging program that allows transmission and reception of single text messages between two Faraday units. Data transfered can be unreliable (no retry if error). This should be a simple and quick stepping stone into getting a reliable data transfer program (auto retry-request).

## Problem Explanation
General requirements:
* Text/Data transfer between two specific units
* Error detection should be implemented
* Error correction should not be implemented
* Restful interfaces provided

### Environment
#### Software 
>Please include the branch or commit if known regarding what software was used if applicable

#### Hardware
Units used during development:
* SN: 13 (No GPS)
* SN: 12 (GPS)

# Supporting Information
>This is a handy spot to post any screenshots or file outputs in order to better support this issue. We want to keep the top of the ticket clutter-free as possible.

kb1lqd, 2017-03-14:
I finished tonight up after pulling in old (but not pulled into master yet in PR's) developer tutorial text messaging program. It needs more cleaning but it is built on simple class objects and works! I modified to meet some basic .INI and other standards we've introduced since. I split the scripts into TX and RX.

BTW we are calling this HERMES as he is the mythical god of the messengers: https://en.wikipedia.org/wiki/Hermes

Below is a screengrab from tonights progress showing RF text transmission between two faraday units using only the basic command application RF forwarding functionality.
[image]

kb1lqd, 2017-03-17:
It should be noted that whenever using the simple command "RF Packet Forward" functionality that every packet must be directed to a specific unit. This doesn't allow general "Chat" with multiple devices.

A specific firmware "app" and UART/RF port must be created to implemented a "broadcast" ability. This is due to all commands being required to directly address the specific unit.

kb1lqd, 2017-03-17:
@kb1lqc 

I created a basic RX Flask server that returns BASE64 encoded received messages from proxy:

[image]

A simple script I wrote `rx_flask.py` using `requests` to decode and display the recieved text message:

[image]

Here is a GIF of me using the simple implementation for receiver.

[image]

kb1lqd, 2017-03-18:
Tonight's update made the receiver functionality even more Flask based and added a dedicated receiver script that queried queuesize and received waiting messages if items were in receiver queue.

TODO:
* Create transmitter module
 * Non-Intuitive since really TX/RX should be in same module and GET/POST interactions largely determine what Faraday does
* How to connect to two different local units for TX/RX full functionality? INI files?

kb1lqd, 2017-03-19:
OK So I renamed the object and device info to generic names that don't imply a TX or RX only nature.

I also confirmed I was able to transmit using FLASK commands into the object!

[image]

I am having trouble though figuring out how to make this modular so that a single file can create a class of two units and can transmit and receive to each other...

73ebd396145bfb444e90d272be0068bf76cd5849

kb1lqd, 2017-03-22:
Arg! I was totally breaking the interface layer by having the application interface directly import a hermes object... That . Is. Bad.

[image]

kb1lqd, 2017-03-22:
I figured it out! I realized I was close but not quite to the architecture that Proxy implements a similar input/output and all I needed to do was make a global dictionary that held TX/RX objects from hermesobject.py with keys related to the respective callsign/nodeid.

I used the `Callsign2COM` function (the name of that function makes no sense in this program) from proxy to parse and create the dictionary of configuration file units to dictionary objects.

[image]

This allows me to use the Hermes Flask server like Proxy and address specific global (dictionary) hermes message objects (tx/rx interfaces) for multiple local units. This allows both a single unit to act as a transciever or multiple units to server transceiver functionality.

The TX and RX Flask user application examples are rough right now but clearly show intended FLASK operation for both TX and RX.

**TX**
[image]

**RX**
[image]

Transmitting shows the properly received "Hello World"

[image]

kb1lqd, 2017-03-23:
@kb1lqc Here's a little proof showing it's Flask :) (using Postman):

[image]

kb1lqc, 2017-03-24:
One way to do this too is to have a worker in the background that saves to an SQLite db and has no knowledge of stations. Your queries with flask provide knowledge. This is how Telemetry tends to work. Not as useful for files but good for messages.

kb1lqd, 2017-04-13:
Merged in #133

closed 2017-04-13 by kb1lqd