FaradayRF/Faraday-Software #132: Proxy Fails To Handle Incorrect Unit Header (count)

kb1lqd, 2017-03-14:
## Summary
When using multiple Faraday units (2) and leaving both the .INI headers [UNIT0] the proxy fails hard. Proxy should gracefully handle this and report the error to the user.

## Problem Explanation
When connecting proxy to multiple local units if the configuration (for 2 units) unit header is [UNIT0] and [UNIT0] instead of [UNIT0] and [UNIT1] the program crashes. Proxy should handle this gracefully and report the error to the use rather than crashing and leaving no debugging information. Otherwise this error should not be allowed to occure if it cannot be reported.

### Environment
#### Software 
Software: Master

#### Hardware
SN: 12, 13

# Supporting Information

Below is an incorrect QTY=2 unit proxy.ini configuration.
[image]

The failure log below when run in a scripting IDE:

[image]

The problem is resolved as expected when configured with the second unit as [UNIT1]

[image]

kb1lqc, 2017-04-30:
Taking a stab at this I implemented proper try/except statement around the `callsign2COM()` function in `proxy.py`. This catches the `ConfigParser.NoSectionError` and exits python.

https://github.com/kb1lqc/Faraday-Software/blob/issue132/faraday/proxy.py#L584

Proper operation

```
C:\Users\Bryce\Documents\GitHub\faradayrf-kb1lqc\Proxy>python proxy.py
2017-04-30 09:41:09,815 - Proxy - INFO - Starting proxy server
2017-04-30 09:41:09,867 - Proxy - INFO - Starting Thread For Unit: KB1LQC-2
2017-04-30 09:41:09,868 - Proxy - INFO - Starting uart_worker thread
2017-04-30 09:41:09,868 - Proxy - INFO - Starting Thread For Unit: KB1LQC-1
2017-04-30 09:41:09,868 - Proxy - INFO - Starting uart_worker thread
```

Exception Operation

```
C:\Users\Bryce\Documents\GitHub\faradayrf-kb1lqc\Proxy>python proxy.py
2017-04-30 09:43:59,013 - Proxy - INFO - Starting proxy server
2017-04-30 09:43:59,015 - root - ERROR - No section: 'UNIT1'
```

kb1lqc, 2017-04-30:
Fix merged into Master! Closing this ticket.

closed 2017-04-30 by kb1lqc