FaradayRF/Faraday-Software #135: Implement Automated Build Testing

kb1lqc, 2017-03-24:
## Summary
We need to become faster at development. Largely it comes down to @kb1lqd and myself (@kb1lqc) to review code prior to pushing into main. This is laborious and often requires actually setting up Faraday hardware to manually move through HITL (Hardware In The Loop) testing 😉. Faraday-Software needs to be able to automatically run and test updates to give reviewers confidence the code is reliable. 

## Problem Explanation
Implementing methods of running python scripts and providing some basic test input would be extremely useful. It would allow us to review code for mistakes but also trust that it was implemented correctly if we understand what the test cases are doing. In a nutshell, if the code is long and complex we can back off on the amount of scrutiny to the code and thoroughly understand the tests so we know that the code is working as a "black box" correctly. This is much more efficient.

### Environment
#### Software 
Faraday-Software repository

#### Hardware
We will likely need either real hardware or at least a test case that mimics real hardware so that code can be run remotely in automated testing.

# Supporting Information
Currently looking at integrating:
- [Travis-CI](https://travis-ci.org/)
- Proxy COM/Fake data application in #108

kb1lqc, 2017-03-24:
Notes:
- http://flask.pocoo.org/docs/0.12/testing/
- https://blog.miguelgrinberg.com/post/the-flask-mega-tutorial-part-vii-unit-testing

reillyeon, 2017-03-24:
A project like this can implement testing at a number of levels but we should probably just start with simple presubmits to ensure that code submitted satisfies basic sanity tests such as importing the FaradayIO library without errors and maybe running a code style checking tool such as flake8. Once we have a presubmit test script hooked up to a service such as Travis CI we can gradually improve the test suite.

For more advanced testing I think we should implement a version of the proxy application that exposes the same REST API but is configurable with a simulated collection of FaradayRF devices that can hear each other with varying levels of packet loss.

kb1lqc, 2017-03-26:
OK this sounds like a great first step @reillyeon I'd appreciate any help on this as I've never really had to do this before. Is there a standard way to do presubmits with Git? I performed a quick Google search and other project presubmits were listed but I can't seem to find a how to from Git or Python on setting up presubmit scripts?

reillyeon, 2017-03-26:
After merging the pull request #136 if you go to https://travis-ci.org/profile/FaradayRF and tick the box for FaradayRF/Faraday-Software their system will start automatically running these tests on all the main branch and all pull requests.

kb1lqc, 2017-04-03:
OK @reillyeon thanks for all the help. I did find [Coveralls](https://coveralls.io/) from another project which seems interesting. Do you see more to do on this Issue Ticket to improve testing or would it be better covered by a specific issue for that style checking?

reillyeon, 2017-04-03:
We should probably close this one since we have the infrastructure set up for automated testing and create issues for particular areas where work is still needed, like finishing up the style fixes and adding tests for particular applications.

kb1lqc, 2017-04-04:
Closing, @reillyeon we should open up issues for specific programs/modules/code that should be tested.

closed 2017-04-04 by kb1lqc