FaradayRF/Faraday-Software #145: Create Simple Telemetry/Command Interface For HAB

kb1lqd, 2017-04-06:
## Summary
The Rose-Hulman amateur radio club is preparing a high-altitude balloon with Faraday on-board and they would like a simple telemetry and command interface. The prior GUI used in https://faradayrf.com/simple-faraday-balloon-launch/ is not currently working with our new modular software. Due to time priority a simpler interface is acceptable.

## Problem Explanation
A simple user interface to:
* Receive and display telemetry live
* Basic telemetry ADC scaling functionality to scale analog sensors
  * Better left for Telemetry?
* Command Faraday for balloon mission

Since proxy and telemetry are Flask based a basic command prompt interface or simple webpage output is acceptable for telemetry viewing.

For commanding a simple command prompt interface is acceptable.

### Environment
#### Software 
Faraday Master branch

#### Hardware

Standard Faraday hardware

# Supporting Information

kb1lqc, 2017-04-06:
Looking at the code we flew with Faraday back in August 2016... it was intimately integrated with Proxy. No bueno. I really don't want to temporarily port that in. We've moved on quite readily.

It will likely be easy to provide an automatically updating jQuery based webpage to see telemetry. No frills. As for commanding, leaving that for command prompt reduces the jQuery and other work to integrate buttons with RESTful commands and therefor would simplify this more.

kb1lqc, 2017-04-07:
I've pushed a few updates to my simpleui branch: https://github.com/kb1lqc/Faraday-Software/tree/simple_web_ui

I now have basic telemetry (subset) code updating to a webpage automatically using JQuery and javascript. This is a good step towards no frills telemetry. This requires the Telemetry CORS update on #148.

Below is a quick screencapture of the interface.

[image]

kb1lqc, 2017-04-09:
Current plan:
- Add commanding support for basic IO commands
- Provide buttons for commands
- Provide simple Bootstrap styling

kb1lqc, 2017-04-10:
Getting there. Boostrap styled UI with some javascript logic to display time and speed nicely as well as detect incorrect input data. Next up simple commanding!

[image]

kb1lqc, 2017-04-14:
Here's the latest UI, a bit rough but coming along. Needs a few of the Firmware and Telemetry issue tickets recently created to really polish up.

[image]

kb1lqc, 2017-04-14:
I've completed my initial run at this user interface. However, for it to fully be useable several issues must be addressed:
- Software #151 
- Software #153
- [Firmware #79](https://github.com/FaradayRF/Faraday-Firmware/issues/72)

In addition to these I'll also need to uncomment and reorganize some HTML code to bring these features online with the UI.

kb1lqc, 2017-04-17:
I updated to properly grab IOstate and merged in a bunch of the latest master changes which changed a lot of uart folders and caused some merging problems with other merges. I've updated the UI to display all known data of interest.

@kb1lqd I cannot confirm the correct orientation of the bits for most IO as the mapping seems to be incorrect in firmware anyways per [Firmware #79](https://github.com/FaradayRF/Faraday-Firmware/issues/72). Any chance we can get this higher priority?

[image]

kb1lqc, 2017-04-18:
Nearly done. Just finished up some styling items and added support for `grunt` checking of javascript and HTML (locally) with node.js using `jslint` and `htmllint`. @reillyeon getting there!

https://github.com/kb1lqc/Faraday-Software/tree/simple_web_ui/Applications/SimpleUI

kb1lqc, 2017-04-19:
OK I updated `jslint` and `htmllint` as best as I could to conform to JQuery formatting per our `CONTRIBUTIONS.MD` file. I'm calling this one done. Eventually we'll revisit this when we automate any of this checking @reillyeon.

I also updated a GPIO bit order error and that fixed a bunch of the GPIO, however, I still can't control or see telemetry that makes sense from GPIO0, GPIO1, or GPIO2:
[image]

I'll probably mark this as a todo update/issue to look into for firmware or software wherever it's located and pull this in on a PR @kb1lqd . That way we get most of the UI into use as we investigate that issue since it adds a lot of value to the project as it is now.

kb1lqd, 2017-04-24:
PR #166 successfully merged and closes this ticket.

closed 2017-04-24 by kb1lqd