FaradayRF/Faraday-Software #147: APRS Readme Erroneously Refers To Passcode Website

kb1lqc, 2017-04-06:
## Summary
APRS [readme.md](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/APRS/readme.md) instructs users to obtain a passcode. The code currently generates the passcode automatically from the callsign rendering this obsolete. This section should be updated to indicate that the passcode is automatically generated.

## Problem Explanation
[generatePasscode()](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/APRS/aprs.py#L636) automatically creates a passcode and returns it to the APRS application to log into APRS-IS with. The documentation should be updated to indicate that the [callsign](https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/APRS/aprs.sample.ini#L6) in aprs.ini is used to generate a passcode when properly configured with the operators callsign.

### Environment
#### Software 
Faraday-Software (Python)

#### Hardware
None, this is a documentation issue

# Supporting Information
[image]

markwiem, 2017-07-13:
Merge indicates that this issue can be closed. True?

kb1lqc, 2017-07-14:
Thanks for the heads-up @markwiem you are correct.

closed 2017-07-14 by kb1lqc