FaradayRF/Faraday-Software #148: Add CORS Support to Telemetry Application

kb1lqc, 2017-04-07:
## Summary
[CORS support](https://enable-cors.org/) is necessary to share data across ports on localhost. [Flask-CORS](https://pypi.python.org/pypi/Flask-Cors) solves this problem easily. We should implement it.

## Problem Explanation
In order for simple local sharing of JSON data across localhost ports the telemetry server must enable CORS support for cross-site access. This is because we jump ports. 

### Environment
#### Software 
Faraday-Software

#### Hardware
N/A

# Supporting Information
If CORS isn't supported by Telemetry then a local javascript function running on port 80 will error when retrieving telemetry data on port 8001:

[image]

This is as easy as implementing the "Simple Usage" example from the [Flask-CORS documentation](https://pypi.python.org/pypi/Flask-Cors).

[image]

jsr38, 2017-04-10:
This looks straightforward.  I'll take it on.  Just brought up Telemetry and Proxy with a fake nodeid and callsign.  I'm trying to check that the CORS headers are correctly generated but as nothing gets sent back to the client I can't see.  Is there a simple way to initialise Telemetry such that it can return something in order for me to verify the headers.  I'm sure I can come up with something - I'm asking in case there is a well-known best way in the absence of Faraday hardware...

kb1lqc, 2017-04-10:
@jsr38 we are not quite yet able to run Faraday software without hardware. Issue #108 pulled in logging functionality to save BASE64 data from a Faraday radio so we can play it back but we haven't implemented the ability to play it back yet. Shortly!

For the time being I'm happy to quickly test. This Issue to add CORS support is literally two lines of code so it should be very straightforward and my description gives you 99% of what you need to know to implement it. Just find the appropriate spots in Telemetry.py to put the import and the CORS wrapper with the flask "app" descriptor and you're set.

This will help get you into the [fork and PR model](https://gist.github.com/Chaser324/ce0505fbed06b947d962) if you're not used to it either.

kb1lqc, 2017-04-11:
Merged in #149. Thanks @jsr38!

closed 2017-04-11 by kb1lqc