FaradayRF/Faraday-Software #152: Enabled Proxy Playback Mode

kb1lqc, 2017-04-11:
## Summary
#108 provided the ability for saving data from hardware into Proxy by means of a SQLite3 database. Now we need to "play" the database file back as if it was real hardware. This will allow us to send a `log.db` file to someone and have them play it as if they had hardware connected. It will also allow automated testing of Telemetry/APRS functionality at the minimum.

## Problem Explanation
Issue #108 created a `log.db` file that we should now allow the configuration of the `Proxy.py` file and config INI to skip setting up a serial port connection and instead open the database and serve each row. At first we should simply serve one row every `rate` seconds which should be set in the INI file. Eventually we may desire to automatically know the rate of the SQLite file. Simplicity is key first.

### Environment
#### Software 
Faraday Software Master branch

#### Hardware
None, please use provided log.db

# Supporting Information
`log.db` SQLite database provided in zip file.
[log.zip](https://github.com/FaradayRF/Faraday-Software/files/913988/log.zip)

jsr38, 2017-04-14:
Right @kb1lqc , I've had a quick look and I propose the following:

- changes to configuration that look like this:  
`[TESTDATABASE]
FILENAME=log.test.db
SCHEMANAME=db.sql

[PROXY]
LOG=0
UNITS=1
TESTMODE=1
TESTRATE=1`

- Where `TESTMODE=1` causes the UART init to be skipped and the DB to be read and inserted on a deque periodically.

- In order to stay consistent with the current Proxy implementation, the playback packets and the deque should be identified as coming from a Station, or callsign-nodeid pair.  I don't think this information is encoded in the packet - please tell me if I'm wrong and where to look if the packet contents can be used to determine this.  I would suspect not in a layered protocol so somehow this info must be provided, perhaps with a `TESTCALLSIGN` and `TESTNODEID` in the config.

- I propose the deque simply overflow if no client is present to read packets.

- Think that use of an asynchronous timer might be better than dealing with the thread directly, although since the uart threads are not going to be started in this situation, it might be simplest to follow the existing pattern and create a `read_testdb_worker` function and sleep.

Let me know if this is along the right lines.

kb1lqc, 2017-04-14:
- Configuration file looks 👍!
- You are correct, the callsign information that refers to which station Proxy is sending data from (which all other applications connection to) is not sent with the packet. Each deque is the value of a dictionary {key:value} where the key is the `callsign-nodeid` of the radio connected via USB.
-Yup, even right now we have a 100 message FIFO buffer that simply drops off the oldest packets one at a time if no one removes them. That's why if you let proxy run for a while and then connect to it there's a flurry of data.
- Yes you might want to consider a separate `read_testdb_worker` that is spun up **instead** of `telemetry_uart_worker`. Especially since #138 is playing with how UART and proxy actually serves data!

jsr38, 2017-04-14:
So maybe
```
[TESTUNIT]
CALLSIGN=replaceme
NODEID=replaceme
```
or something else?

kb1lqc, 2017-04-14:
@jsr38 that sounds good to me

kb1lqc, 2017-04-16:
Thanks for the update @jsr38 ! Closing this ticket as it's main purpose is now complete.

closed 2017-04-16 by kb1lqc