FaradayRF/Faraday-Software #153: Remote Commanding of MOSFET/HAB Timer Does Not Work

kb1lqc, 2017-04-14:
## Summary
During #145 development it was observed that MOSFET and HAB Timer commands to remote Faraday stations does not work. These should be working.

## Problem Explanation
The following commands do not produce the expected results:
- [HAB Cut-down Event](https://github.com/kb1lqc/Faraday-Software/blob/simple_web_ui/Applications/SimpleUI/simpleui.py#L320)
- [Reset HAB Cut-down Timer](https://github.com/kb1lqc/Faraday-Software/blob/simple_web_ui/Applications/SimpleUI/simpleui.py#L324)
- [Disable HAB Cut-down Timer](https://github.com/kb1lqc/Faraday-Software/blob/simple_web_ui/Applications/SimpleUI/simpleui.py#L328)
- [HAB Reset Cut-down Idle](https://github.com/kb1lqc/Faraday-Software/blob/simple_web_ui/Applications/SimpleUI/simpleui.py#L332)

### Environment
#### Software 
Software SimpleUI Branch application: https://github.com/kb1lqc/Faraday-Software/tree/simple_web_ui/Applications/SimpleUI

#### Hardware
>Let us know about what hardware you are using if applicable. This includes:
- Using Windows 10 on a Microsoft Surface Pro 4
- Normal Rev D1 Faraday radios

# Supporting Information
None

kb1lqd, 2017-04-18:
Verified that local MOSFET cutdown works as intended:
[image]

kb1lqd, 2017-04-18:
Observing a local unit sending the RF command it is obviuos that there is a missing frame in the RF packet being sent. The callsign "K" is in the command position where RF remote command `9` should be. 

[image]

kb1lqd, 2017-04-18:
OK I think I fixed it, @kb1lqc you deviated from your working implementations of remote commanding to do the MOSFET actions as if they were "special" when they are in fact not. Adding the local command `9` to send the packet over RF to a remote device looks to have solved the issue. I haven't tried this on a real remote unit (I don't have gear to test this MOSFET at the moment):

Modifying:

`command = faraday_cmd.CommandRemoteHABActivateCutdownEvent(remotecallsign, remotenodeid)`

to

`command = faraday_cmd.CommandLocal(9, faraday_cmd.CommandRemoteHABActivateCutdownEvent(remotecallsign, remotenodeid))
`
Looks to have created the proper packet as I'd expect, @kb1lqc please confirm this solves the issue and roll into all remote command as needed (looks like only the MOSFET ones are missing it).

[image]

[image]

kb1lqc, 2017-04-18:
Ok I'll test. Wow yeah I just read through code and found what appeared to be correct since the remote MOSFET wasn't in any examples I could find. I'll test.

P.S. Instead of images (or in addition to), you can link directly to the line 😉

kb1lqc, 2017-04-18:
OK most seem to work @kb1lqd. I think I need to check the Idle timer and actually look at the MOSFET resistance during firing with a multimeter. However, just by states I can tell the command of HAB timers seems to be working and the MOSFET firing also seems to work!

[image]

kb1lqc, 2017-04-18:
Yes @kb1lqd this is odd or at least not what I expect.

When the MOSFET hasn't fired yet but timer is counting and I send an Idle... nothing happens.
[image]

Yet when I have fired the MOSFET (STATE = 255) and set Idle the Cut-Down State returns to 0 without anything else changing.
[image]

kb1lqd, 2017-04-18:
That's exactly what I'd expect. Idle set is only used to set the state
machine into idle 0, most likely from idle 255 post firing. If your in 0
and command idle 0 you'll do nothing...

This is so you can reset from post fire 255 and catch a future firing
without needing to catch telem as it happens.

kb1lqc, 2017-04-19:
Closing this as @kb1lqd pointed out I had used the commands incorrectly implemented. This is fixed in the branch the issue was found in.

https://github.com/kb1lqc/Faraday-Software/tree/simple_web_ui

Specifically the commit it was fixed in:
407f8bec3d461c158ad30dc17d1545515f7a1685

closed 2017-04-19 by kb1lqc