FaradayRF/Faraday-Software #163: Fix up Proxy to conform to contributions style guidelines

jsr38, 2017-04-17:
## Summary
Ensure that code in proxy.py conforms to code contribution style guidelines which includes but is not limited to function and variable naming

## Problem Explanation
This is mostly about naming schemes so far as I can tell.  Proxy was introduced before the guidelines existed and subsequent changes have followed the existing conventions for consistency.  This ticket is about doing a wholesale naming and other style cleanup of proxy.py to  bring it in line with later contributions.

### Environment
#### Software 
Faraday Software master branch

#### Hardware
Not applicable

# Supporting Information
[CONTRIBUTING.md](CONTRIBUTING.md) and [PEP8](https://www.python.org/dev/peps/pep-0257/)

kb1lqc, 2017-04-18:
I assigned this to you @jsr38 since you created it. Let me know if you'd rather not!

jsr38, 2017-04-18:
No problem.  Might not get done today but I'll take it...

kb1lqc, 2017-04-25:
Hey @jsr38 still thinking about giving this one a shot?