FaradayRF/Faraday-Software #175: Links to Proxy Folder Broken

kb1lqc, 2017-04-25:
## Summary
We recently fixed an error in #172 which resulted in a secondary issue of our documentation linking to the 'proxy' folder no longer working.  We need to update the links referring to the folder to use "Proxy" instead. This is project-wide and examples of a few updates can be seen in #173  and #174.

## Problem Explanation
Likely problems spots include:
- Application readme.md files
- tutorial .md files

### Environment
#### Software 
Documentation/GitHub

#### Hardware
N/A

# Supporting Information
N/A

morrme, 2017-04-25:
Hello again! I can help with this.

morrme, 2017-04-26:
I have only found 2 references that hadn't been fixed in this repo, would there be other references in the other repositiories that should be checked ?

kb1lqc, 2017-04-26:
Not in other Repo's that I'm aware of. If there are only two then that's awesome!

Did you read through the /tutorials/start and the /applications documents to check for the links?

morrme, 2017-04-26:
@kb1lqc yes.  Just one comment and one link needed corrections.

( https://github.com/FaradayRF/Faraday-Software/compare/master...morrme:issue-175?expand=1 )

kb1lqc, 2017-04-26:
Nice! I guess it’s time to create a PR then. Thanks for checking.

 

( https://github.com/FaradayRF/Faraday-Software/compare/master...morrme:issue-175?expand=1 )

PR #179 (2017-04-26): Changed Proxy references per issue #175

Fixes: #175
closed 2017-04-26 by kb1lqc