FaradayRF/Faraday-Software #176: Applications require additional modules.

Kacela, 2017-04-25:
## Summary
As of April 24th build, at least 2 of the core applications require that additional modules/dependencies  be installed before they will function.

## Problem Explanation
With the latest (April 24) Faraday-Software build, the applications telemetry.py and simpleui.py will not function until additional modules are imported.  Telemetry.py requires the module 'flask_cors', and SimpleUI.py requires 'flask_bootstrap'.

Branch: master
Merge pull request #166 from kb1lqc/simple_web_ui 

#### Hardware
Faraday radios SN 3 & 51, running with Windows 10

# Supporting Information
Telemetry.py returns:
Traceback (most recent call last): 
  File "telemetry.py", line 26, in <module> from flask_cors import CORS ImportError: No module named flask_cors'

SimpleUI.py returns a similar failure message about missing flask_bootstrap

Both issues easy to overcome, though importing flask_bootsrap seems a bit "bloaty"

kb1lqc, 2017-04-26:
Can confirm this works fine on my master builds with CORS support:
[image]

Uninstalled:
- Flask-Cors-3.0.2
- Flask-Bootstrap

Proxy runs fine but Telemetry fails:
[image]

Can also confirm that commenting out the Boostrap code allows SimpleUI to run without need for the Flask-Bootstrap module.
Import: https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/SimpleUI/simpleui.py#L19
Bootstrap(App): https://github.com/FaradayRF/Faraday-Software/blob/master/Applications/SimpleUI/simpleui.py#L41

The Flask-Bootstrap module was accidentally left in. I ended up just leaving the real Bootstrap Javascript library in there instead of using the Python module but forgot to remove the code.

kb1lqc, 2017-04-26:
I can confirm however that the `Flask-CORS` module is indeed in the `requirements.txt` file:

https://github.com/FaradayRF/Faraday-Software/blob/master/requirements.txt#L7

@Kacela you should make sure to run `pip install -r requirements.txt` from command line in the main FaradayRF folder This should install Flask-CORS and any other necessary modules. You can also check with a `pip freeze` to see if it was installed.

kb1lqc, 2017-04-28:
@Kacela did this fix your issue?

Kacela, 2017-04-29:
Not seeing a response I posted 5 hours ago, so copying and pasting.
BLUF: Issue resolved to my satisfaction. 

<paste> Yes, issues resolved.  I performed a clean Python install as well as installed a new copy of Faraday-Software. I can confirm that I am no longer required to install Flask_Bootstrap nor Flask_CORS individually.  I can also confirm that Flask_CORS is now in the requirements.txt file, and Flask_CORS is properly brought in after issuing pip install -r requirements.txt (What you refer to as the main FaradayRF folder for me is actually C:\FaradayRF\Software\ ). Proxy, Telemetry, APRS, and SimpleUI are all operating nominally for me now, with no surprises on invoking them. </paste>

closed 2017-04-29 by Kacela