FaradayRF/Faraday-Software #194: Inconsistent use of "while True:" and "while 1:"

el-iso, 2017-04-30:
## Summary
faraday/telemetry.py and faraday/aprs.py both use "while true:" several times for infinite loops. Where as faraday/proxy.py uses "while 1:" for its main infinite loop. 

## Problem Explanation
Evidently, in python 2.7 "True" and "1" work slightly differently from a logical perspective. Referring to [this Stack Overflow post](http://stackoverflow.com/questions/3815359/while-1-vs-for-whiletrue-why-is-there-a-difference) and [this page in the Python docs](https://docs.python.org/3/library/constants.html#True), in Python 2.7 True is a "Built-in Constant" that equals "1". 

Because of this, 
* Using "while True:" creates more byte code than using "while 1:" so its slightly less efficient
* The value of "True" can be reassigned

It seems like usage of "while True:" through out the faraday software should be changed to "while 1:" for the sake of consistency a to reduce computation costs (especially since some of these loops run a lot).

### Environment
#### Software 
The master branch as of April 30th. I have mainly found this in faraday/aprs.py and faraday/telemetry.py though it could exist elsewhere.

#### Hardware
This is a software issure.

# Supporting Information
Again, [this Stack Overflow post shows the differences in byte code, and the third response has a computation comparison](http://stackoverflow.com/questions/3815359/while-1-vs-for-whiletrue-why-is-there-a-difference) 
and [this page in the Python docs page talks about built in constants in python](https://docs.python.org/3/library/constants.html#True)

reillyeon, 2017-04-30:
It's fixed in Python 3. Can we just switch? ;P

Right now our main loops have sleeps in them because we don't actually want them to run really fast because that burns CPU. This code should really be rewritten to use the correct OS methods to cause the process to sleep until new data is available instead. For this reason I don't think the performance cost of `while True` vs. `while 1` is worth chasing. The benchmarks above only show a difference because the added overhead of checking against `True` takes more opcodes than the code within the loop itself which is certainly not the case for us.

As a style thing I think that Python clearly wants you to use `True` to represent a value that will always be truthy. `1` is also a truthy value so it works just as well but to me it takes a little bit more cognitive effort to remember than non-zero values are truthy. As the project's self-appointed arbiter of style I think we should stick with `while True` while keeping in mind that any place we have this construct is also a bug because busy loops worse for power consumption than a few extra opcodes to look up a global.

el-iso, 2017-04-30:
@reillyeon Considering style, should usage of "while 1:" then be changed to "while true:" 

Also, "while(True)" is used in some places while "while True:" is used in others. Should that be standardized and is one more favorable than the other? I tried to find information about that in the python style guide but I couldn't find anything.

kb1lqc, 2017-04-30:
@reillyeon well said. When we started Faraday several iterations of software (trust me, NOTHING like it is today 😟 ) this was several years ago and Python 2.7 was still king for most modules. My understanding is that Python 3 is extremely well supported these days. Is the discussion of moving to Python 3 worth having right now or more of distraction?

reillyeon, 2017-04-30:
@el-iso Patches to improve consistency and remove unnecessary parenthesis are welcome.

@kb1lqc We should upgrade to Python 3 but since we don't have enough automated test coverage we probably don't want to try supporting both at once. It's possible but means manually testing every patch twice. If you want to declare a Python 3 flag day I'd support it but don't want it to interfere with other projects you're working on that are higher priority.

kb1lqc, 2017-04-30:
@reillyeon makes sense. Maybe we should add a issue ticket to not forget. On the other hand it's pretty blatantly not Python 3 😆 . I agree @el-iso feel free to submit a PR to update styling.

morrme, 2017-05-04:
I am always happy to help with cleanup tasks as well. :smiley:

kb1lqc, 2017-05-04:
@el-iso  and @morrme why not quickly chat and see who wants to knock this one out?

el-iso, 2017-05-04:
@morrme you're welcome to take this one if you want. I'm in the thick of final projects and exams so it's going to be a couple days before I get to it. So, go right ahead if you'd like!

morrme, 2017-05-04:
@el-iso @kb1lqc Thanks! Is the preferred style while(True) or while True ? And should the "while 1:" lines be changed as well? (cc: @reillyeon )

kb1lqc, 2017-05-04:
@morrme We prefer `while True`. Also, yes the `while 1` lines should be changed.

morrme, 2017-05-04:
@kb1lqc done.

kb1lqc, 2017-05-04:
I'll try to get to this in the next few days. I appreciate it! Busy with #191 up to the gills right now.

PR #198 (2017-05-06): standardize infinite loops
Fixes: #194
closed 2017-05-06 by reillyeon