FaradayRF/Faraday-Software #216: Pip Installation faraday-telemetry Not Creating telemetrydb.sql In etc\faraday Folder

kb1lqd, 2017-06-28:
## Summary
I started from a fresh `...\etc\faraday` folder and pip installed faraday. I found that proxy ran successfully but when trying to initialize and run faraday-telemetry the program could not find the needed SQL file and crashed.

## Problem Explanation
I've pasted an image of the error below in supporting information. 

I was able to successfully run telemetry after manually copying the `db.sql` file into the `..\etc\faraday\` folder and renaming it to `telemetrydb.sql` from the master github repository.

### Environment
#### Software 
Pip installed faraday version information.
```
C:\Users\Brent>pip show faraday
Name: faraday
Version: 0.0.1013.dev1040
Summary: FaradayRF amateur radio open source software
Home-page: https://github.com/FaradayRF/Faraday-Software
Author: FaradayRF
Author-email: Support@FaradayRF.com
License: GPLv3
Location: c:\python27\lib\site-packages
Requires: pytest-flake8, six, py, requests, Flask-Cors, packaging, pyflakes, MarkupSafe, itsdangerous, appdirs, pyserial, pycodestyle, Werkzeug, Flask, pyparsing, Jinja2, click, flake8, pytest, configparser, mccabe, enum34
```
Master github software.

#### Hardware
Faraday REV D1 units.

# Supporting Information

Error received:
[image]

Github repo telemetry db.sql file, I renamed this to telemetrydb.sql and moved into the etc/faraday folder.

[image]

The etc/faraday folder. Note that the `db.sql` file is for proxy not telemetry.

[image]

Succesfully running telemetry after manually copying over telemetrydb.sql file.

[image]

kb1lqc, 2017-06-29:
Installing with `pip install --no-cache-dir faraday` results in this working for me just fine. However I just realized that's because I haven't cleared my `telemetry.db` which will persist between uninstalls apparently.

[image]

This results in the new installation not having to create a database and therefore not needing the schema. Deleting `telemetry.db` results in the error since I now run the code that creates a database from the `telemetrydb.sql` file:

[image]

So confirmed...

# The Fix
Simply adding `etc/faraday/telemetrydb.sql` to `setup.cfg` is all that's needed. This tells setuptools to copy the local `telemetrydb.sql` into the main `/etc/faraday` folder. Testing this with a locally created wheel distribution file I successfully installed faraday and obtained the needed file.

[image]

Starting with a clean `.faraday/` folder:

[image]

Launching `faraday-telemetry` worked fine:
[image]

You can see that the `telemetry.db` log file was created successfully!
[image]

Fixed in PR#217

kb1lqd, 2017-07-08:
#217 closes this.

closed 2017-07-08 by kb1lqd