FaradayRF/Faraday-Software #225: Create Bash/Batch Script to Automate Application Initialization

kb1lqc, 2017-07-04:
## Summary
Per @kb1lqd's [comment](https://github.com/FaradayRF/Faraday-Software/pull/222#discussion_r125403552) this would be a fun and easy project to add automation to the installation of Faraday software. Many program files need to be initialized with `--init-config` and similar commands which can be scripted. Additionally many need programmed callsign values which can be automated too.

## Problem Explanation
We should strive to simplify Faraday operation as much as possible. Now that we can run Faraday software from the command line we can also script many common actions. Initialization and configuration of our software is a lot less painful than it was but this doesn't mean there's not more we can do. Creating a simple `install.sh` or `install.bat` script that performs the monotonous portion of setting up Faraday would be incredibly beneficial to the project.

### Environment
#### Software 
Master Faraday software
#### Hardware
N/A

# Supporting Information
N/A

reillyeon, 2017-07-04:
Maybe a `faraday-setup` command that does the various initialization steps with prompts for important configuration parameters like station callsign.

kb1lqd, 2017-07-05:
I'ver created the following BATCH scripts. I expect to create a branch and PR for them. There is likely some cleaning up for them but they are working well with the latest `--start` PR merge master!

I'm sure this structure can easily be ported to BASH as well.

[faraday-bat.zip](https://github.com/FaradayRF/Faraday-Software/files/1126100/faraday-bat.zip)

@reillyeon  @el-iso  @kb1lqc @hdkmike want to review/improve/comment on them?

hdkmike, 2017-07-06:
@kb1lqd I'll definitely take a look at this tomorrow.

hdkmike, 2017-07-07:
@kb1lqd These look pretty handy! Nice work! Other than anonymizing the ports and your callsign, they look fairly good to get up and running quickly. Unfortunately, I haven't run them yet but look forward to trying them soon! Let me know when you have a PR up.

kb1lqd, 2017-07-07:
Will do! I've been using and improving then as I review Bryce's PR's and
even mine adding the generic data server. This command line interfaces for
Faraday are very very slick! One command run to open and configure a ton of
stuff.

el-iso, 2017-07-09:
@kb1lqd Just going through the new setup for the first time and my first thought was that some scripts like this would be very useful! I'm going to read up on batch scripts (I'm mostly ignorant at this point) and try to read over them on a higher level..

kb1lqd, 2017-07-09:
Hey I started working a python version of these using subprocess call
function to be cross platform. Thanks for using these though! Yeah they are
very helpful IMHO.

kb1lqd, 2017-07-10:
Hey @el-iso @reillyeon  @kb1lqc @hdkmike I created a branch in my fork that holds "nearly" platform independent simple scripts that make using Faraday from command line easy.

https://github.com/kb1lqd/Faraday-Software/tree/simple_scripts/simple_scripts

Let me know what you think!

kb1lqd, 2017-07-15:
#246 closed this.

closed 2017-07-15 by kb1lqd