FaradayRF/Faraday-Software #226: Ensure telemetrydb.sql is Properly Copied on Fresh Install of Faraday Software

kb1lqc, 2017-07-04:
## Summary
@kb1lqd is still [reporting](https://github.com/FaradayRF/Faraday-Software/pull/222#discussion_r125405271) that this issue exists. It's being reported on line [626 of telemetry.py](https://github.com/FaradayRF/Faraday-Software/blob/master/faraday/telemetry.py#L626) which should fail gracefully with an exception that is caught. Additionally, it appears that the [schemaname is correct](https://github.com/FaradayRF/Faraday-Software/blob/master/etc/faraday/telemetry.sample.ini#L7) in telemetry.sample.ini... 

## Problem Explanation
N/A

### Environment
#### Software 
Faraday Master Branch and issue220 of kb1lqc fork

#### Hardware
N/A

# Supporting Information

[image]

kb1lqc, 2017-07-04:
Used latest Master Branch with `pip install .` resulting in:

```
$ pip show faraday
Name: faraday
Version: 0.0.1014.dev1121
Summary: FaradayRF amateur radio open source software
Home-page: https://github.com/FaradayRF/Faraday-Software
Author: FaradayRF
Author-email: Support@FaradayRF.com
License: GPLv3
Location: c:\python27\lib\site-packages
Requires: enum34, itsdangerous, pycodestyle, pyserial, MarkupSafe, pytest, Jinja2, click, requests, flake8, pyparsing, py, six, Flask, Werkzeug, packaging, mccabe, configparser, appdirs, pyflakes, Flask-Cors, pytest-flake8
```

Telemetry worked fine. Cannot reproduce.

kb1lqd, 2017-07-05:
Verified it has been working for me again on latest installations, I also cannot reproduce so I am closing this. Hopefully this was a fluke or user-error on my part :|

closed 2017-07-05 by kb1lqd