FaradayRF/Faraday-Software #241: Pip Install Issue - Version RPi

kb1lqd, 2017-07-07:
## Summary
While trying to install Faraday software on my RPi3 through PIP (PyPi) I encountered issues with Faraday not being found in pypi due to incorrect version.

I used `pip install Faraday` and it failed.

## Problem Explanation

I was able to install using `sudo pip install Faraday==0.0.1015.dev1222` (I had to use sudo...)

But I did get odd operation/failure of PIP after installation, I have use pip or uninstall things anymore...

### Environment
#### Software 
PyPi installation of Faraday 0.0.1015.dev1222

#### Hardware
Raspberry Pi 3.

# Supporting Information

Failure terminal:
[image]

Pip now failure during any use after installation of faraday:

[image]

kb1lqd, 2017-07-08:
https://stackoverflow.com/questions/27341064/how-do-i-fix-importerror-cannot-import-name-incompleteread

kb1lqd, 2017-07-08:
I seem to have found a directory issue with the Pypi installation on linus @kb1lqc 

[image]

Faraday isn't finding the .ini files because they are not here @kb1lqc

kb1lqd, 2017-07-08:
Installing in editable mode from repo:

``` bash
kb1lqd@raspberrypi-kb1lqd:~/Documents/Git_Faraday_Software/master/Faraday-Softwa                     re $ sudo pip install -e .
Obtaining file:///home/kb1lqd/Documents/Git_Faraday_Software/master/Faraday-Soft                     ware
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/kb1lqd/Documents/Git_Faraday_Software/master/Faraday-Software/                     setup.py", line 7, in <module>
        pbr=True,
      File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 262, in _                     _init__
        self.fetch_build_eggs(attrs['setup_requires'])
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 287, in f                     etch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 631, in res                     olve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 874, in bes                     t_match
        return self.obtain(req, installer)
      File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 886, in obt                     ain
        return installer(requirement)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 338, in f                     etch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py"                     , line 636, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py"                     , line 666, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py"                     , line 856, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py"                     , line 1078, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py"                     , line 1063, in run_setup
        run_setup(setup_script, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 68, in                      run_setup
        DirectorySandbox(setup_dir).run(runner)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 114, i                     n run
        return func()
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 67, in                      runner
        _execfile(setup_script, ns)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 43, in                      _execfile
        exec(code, globals, locals)
      File "/tmp/easy_install-ATgFvi/setuptools-36.0.1/setup.py", line 187, in <                     module>

      File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 266, in _                     _init__
        _Distribution.__init__(self,attrs)
      File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
        self.finalize_options()
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 301, in f                     inalize_options
        ep.load()(self, ep.name, value)
      File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2195, in lo                     ad
        raise ImportError("%r has no %r attribute" % (entry, attr))
    ImportError: <module 'setuptools.dist' from '/usr/lib/python2.7/dist-package                     s/setuptools/dist.pyc'> has no 'check_specifier' attribute

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/kb1lqd/Docu                     ments/Git_Faraday_Software/master/Faraday-Software/
kb1lqd@raspberrypi-kb1lqd:~/Documents/Git_Faraday_Software/master/Faraday-Softwa                     re $

```

kb1lqd, 2017-07-08:
Wow It was because I had a wicked old `setuptools`:

kb1lqd@raspberrypi-kb1lqd:~/Documents/Git_Faraday_Software/master/Faraday-Software $ sudo pip install setuptools --upgrade
Collecting setuptools
  Downloading setuptools-36.0.1-py2.py3-none-any.whl (476kB)
    100% |████████████████████████████████| 481kB 498kB/s
Installing collected packages: setuptools
  Found existing installation: setuptools 5.5.1
    Uninstalling setuptools-5.5.1:
      Successfully uninstalled setuptools-5.5.1
Successfully installed setuptools-36.0.1

Installed!

```bash
kb1lqd@raspberrypi-kb1lqd:~/Documents/Git_Faraday_Software/master/Faraday-Software $ sudo pip install -e .
Obtaining file:///home/kb1lqd/Documents/Git_Faraday_Software/master/Faraday-Software
Requirement already satisfied: appdirs==1.4.3 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: click==6.7 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: configparser==3.5.0 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: enum34==1.1.6 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: flake8==3.3.0 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: Flask==0.12 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: Flask-Cors==3.0.2 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: itsdangerous==0.24 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: Jinja2==2.9.4 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: MarkupSafe==0.23 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: mccabe==0.6.1 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: packaging==16.8 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: py==1.4.33 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: pycodestyle==2.3.1 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: pyflakes==1.5.0 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: pyparsing==2.2.0 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: pyserial==3.2.1 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: pytest==3.0.7 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: pytest-flake8==0.8.1 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: requests==2.12.5 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: six==1.10.0 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: Werkzeug==0.11.15 in /usr/local/lib/python2.7/dist-packages (from faraday==0.0.1015.dev1223)
Requirement already satisfied: setuptools in /usr/local/lib/python2.7/dist-packages (from pytest==3.0.7->faraday==0.0.1015.dev1223)
Installing collected packages: faraday
  Running setup.py develop for faraday
Successfully installed faraday
kb1lqd@raspberrypi-kb1lqd:~/Documents/Git_Faraday_Software/master/Faraday-Software $
```

kb1lqd, 2017-07-08:
I got proxy and data running after installing in editable mode!

```bash
Enter Message: This is a test
Transmitting [0]: S0IxTFFEAv4OVGhpcyBpcyBhIHRlc3QAAAAAAAAAAAAAAAAAAAAAAA==
Enter Message:
```

I couldn't get bash scripts to work... and I needed to use `tmux` for multiple windows and had to deal with the processes not dieing and manually finding/killing them...

A good effort to make linux operation better should be in store.

kb1lqc, 2017-09-10:
@kb1lqd did you try `faraday-telemetry` on the Raspberry Pi yet?