FaradayRF/Faraday-Software #243: Linux - How to connect proxy to /dev/ttyUSBx without sudo privledges

kb1lqd, 2017-07-11:
## Summary
Previously for linux users it was not known (in general/documentation) how to start proxy without running sudo (or with root privileges). This is annoying at best and also makes automation scripts difficult. This issue ticket is to be used to explain how to open proxy connections to Faraday units without root privileges.

## Problem Explanation
The user account running the proxy program must be added to the `dialout` group. Doing so will allow the user to open proxy and connect to a unit without the "permission denied" error.

This information should:
* Be included in documentation
* Be stated in relevant scripts/programs opening proxy to allow easier debugging.

### Environment
#### Software 
* Faraday
  * Master
* Linux
  * Ubuntu 16.04 LTS

#### Hardware
N/A

# Supporting Information

Running the following command in terminal will open a new terminal window running faraday-proxy:

`gnome-terminal -x faraday-proxy --number 1 --start`

If the user is not in the dialout user group this will fail to connect stating 'Permission Denied'. The user is left to either use sudo every time proxy is to be run or add their user name to the linux `dialout` group.

Simply adding the user to the `dialout` group and logging out and back in will solve the problem.

`sudo usermod -a -G dialout $USER`

## References
* https://askubuntu.com/questions/133235/how-do-i-allow-non-root-access-to-ttyusb0-on-12-04

kb1lqd, 2017-07-11:
@jbemenheiser @hdkmike @el-iso @reillyeon may be interested in this.

kb1lqc, 2017-10-06:
We should probably document this better with #270 as well. I definitely ran into this + some. I also noticed on the latest Raspbian I had to install as sudo too or else no command line arguments would be installed :/

herrold, 2017-10-06:
In the Subject the term: proxy  is a term of art in computing

'Permit non-privileged user access' is perhaps more useful so that the search engines mis-index

kb1lqc, 2017-10-07:
@herrold proxy is the name of our software that interfaces the Faraday hardware & network.