FaradayRF/Faraday-Software #260: Integrate useful HAB sensors into Faraday.

kb1lqd, 2017-09-01:
Purchase and/or design sensor board(s) that integrate into Faraday's ADC's or SPI interface for use on the HAB mission. Educational documentation should be created at a system level to show how Faraday integrates with sensors.

kb1lqc, 2017-09-04:
Sounds good. You sure we want to create conditioning boards for 4-20mA or 0-5V? I take it this is the lesser of the evils for available sensor products that are mostly off the shelf?

kb1lqd, 2017-09-04:
Well @kb1lqc I'd prefer 0-5V scaling to 0V to VREF as it's simpler. We could try to find all I2C sensors too but I'm a little worried on getting I2C going... just from a gotcha's and such standpoint.

kb1lqc, 2017-09-04:
OK @kb1lqd, at what point does interfacing I2C just buy us a big bang for the buck? Adding on conditioning hardware is a tad much if there are better and more convenient options in I2C.

kb1lqd, 2017-09-04:
OK well let me see what I can find.

kb1lqd, 2017-09-05:
@kb1lqc I recall my largest concern about the I2C:

We'll be using the USCI_Bx bus and need to enable I2C mode. USCI_Bx is IN USE for the SRAM in SPI mode so we'll need to share with the sensors.

So we'll need to swap between SPI and I2C and while I2C is in use we can't use the SRAM.

kb1lqd, 2017-09-06:
@kb1lqc Are we planned out enough here to pull the trigger on buying sensors?

kb1lqc, 2017-09-06:
@kb1lqd what about the [DS28E17](https://datasheets.maximintegrated.com/en/ds/DS28E17.pdf) from Maxim IC for 1 wire to I2C? I bet there are breakout boards for this.

kb1lqc, 2017-09-06:
Possibly useful cheap board for 1-wire to I2C: http://modtronix.com/im1wp-tbb.html
more expensive: https://www.tindie.com/products/closedcube/ds2482-800-i2c-to-8-channel-1-wire-breakout-board/

kb1lqc, 2017-09-06:
Also @kb1lqd these Blue Robotics cable penetrators look awesome!

http://bluerobotics.com/store/connectors/penetrator-10-25-a/
[image]
[image]

kb1lqc, 2017-09-06:
OK looking at the sensors I have a few comments @kb1lqd 
* We're initially using APRS-IS (at this time) to transfer data onto the internet which only has allocations for five telemetry channels
* We do save our own data and have it all but this is something to consider for the first buoy
* currently we use the internal 2.41V reference which isn't terribly good (1.5% I believe). Is there actually a 2.5V internal reference? Do we want an external reference?
* For the 0-2V signals I may just say don't buffer them, is it worth full dynamic range of 2.41V to add that many more op-amps? I supposed if we have an external board with one op-amp then what is a few more...
* I am actually not sure we want to use 0-5V. If my memory serves me well it can be a pain in the butt to scale down from 0-5V to 0-2.41V. Probably easy with a voltage divider and op-amp buffers but we could easily see more noise here. I wonder if 4-20mA works better here. It's more power but about the same complexity if we are buffering (current sense amplifier instead of buffer amp). It's worth thinking more about.

kb1lqc, 2017-09-06:
Overall I have some comments on general sensors
* We probably shouldn't go overboard on the pure number of sensor channels. Can you put some thought into why we would want each channel? Maybe compare against other buoys. We should focus on providing the minimum in useful design for this first one. Also check out [this project](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3444120/) which has some good ideas.

kb1lqc, 2017-09-06:
@kb1lqd check out the NOAA buoys: http://tao.ndbc.noaa.gov/proj_overview/sensors_ndbc.shtml

You can actually search some of their sensors. Pricey (I mean... it probably costs a LOT in fuel and labor to send a boat out to put the expensive buoy in these places anyways!).

*[Wind Sensor](http://www.youngusa.com/products/7/5.html)
* [Air Temperature](https://www.rotronic.com/en-us/mp10xa-mp40xa.html)

Not sure how useful but it's interesting.

kb1lqc, 2017-09-06:
Oh neat, here are some more specifications for buoys

[Drifter](http://www.jcommops.org/FTPRoot/DBCP/Documents/standards/calibration/Min_spec_drift.pdf)
[Moored](http://www.jcommops.org/FTPRoot/DBCP/Documents/standards/calibration/Min_spec_moored.pdf)

kb1lqd, 2017-09-07:
I Soldered jumpers to the pins on Faraday to allow bit-banged I2C. Note that I needed to use AD2 as a GPIO because it is the only GPIO port left with interrupt capability externally.

[image]

I also hooked up the RPI to I2C for both basestation I2C ADC to sense shutdown voltage and if it is needed to just use for I2C sensors.

[image]

@kb1lqc will be interested.

kb1lqd, 2017-09-08:
@kb1lqc  (Read all, especially near bottom)

I found http://www.ti.com/lit/an/slaa703/slaa703.pdf which the ZIP download has usable code it seems. This does need a timer that I may not have right now. 

My setup took me a while after getting home today and now code composer isn't seeing Faraday... I may try again not at 11PM.

I'm wondering how long I give tying to get Faraday to bit bang I2C versus just use the working Rpi I2C? I know that keeping the smart modem capability is ideal but wondering how much, at least for now to keep things moving.

**Faraday Base I2C**
* Port I2C bit bang code to Faraday
* Add I2C data gathering code to Faraday for each sensor
* Add I2C data to telemetry packets
   * Likely going to need to make multiple telemetry packets to fit it all so a good restructuring of telemetry is a good idea

**Rpi Based I2C**
* Create scripts to gather I2C sensor data
* Create scripts to EITHER
  * Send I2C data to faraday and have faraday create packets
  * _Directly create packets in RPi_ and only have Faraday transmit them

I think that we might want to consider if moving to a much more scale-able telemetry creation implementation is a good idea and base it heavily on RPi? Maybe leave the ability for Faraday's base telemetry but add/modify to easily scale into N amount of data from a RPi through proxy that is sent over RF. 

Maybe Faraday abstracts a "telemetry" packet and type like it does now but add the functionality to sends new packets from a remote computer source (RPi).

Currently:
Telem Packet #1 = System data (i.e. frequency)
Telem Packet #2 = Debug information
Telem Packet #3 = Faraday Sensor Telemetry (i.e ADC/GPS/etc...)

Maybe the easiest to do is just add additional packet types (i.e. #4) and let them come from the RPi itself and Faraday simply wraps it and sends is as a telemetry packet?

If we do this do we want to both with I2C on Faraday right now and just have RPi do it? This leaves any basic GPIO/ADC's for Faraday but any hardcore sensors go to I2C on RPi. 

This may spur a small side project on Faraday Software for a "Sensors/Telemetry Generating Program" that runs with Proxy to gather and create generic telemetry...

kb1lqc, 2017-09-08:
I'll respond in the morning @kb1lqd but it occured to me. What about using I2C SRAM instead of switching between SPI and I2C? This seems to solve the crossing of the stream so-to-speak.

kb1lqc, 2017-09-08:
You should at least give I2C a shot on Faraday. It makes sense to not have to use the RPI to do this. In the future we may want to delegate much to the RPI I agree.

Make sure to consider the options that would make this quicker. What do you think about using an I2C to SPI converter and keeping everything SPI to the CC430 and all external I2C goes through the converter? the CP2120 is an example and here is a [dev board](https://www.silabs.com/products/development-tools/interface/cp2120ek-evaluation-kit). Building a simple converter board or finding a low-cost alternative could help us immensely. Let's not brute force this is we don't have to.

Will a SPI to I2C converter help a lot?

kb1lqc, 2017-10-06:
Sensors in transit @kb1lqd:

* https://www.adafruit.com/product/1733
* https://www.adafruit.com/product/1777
* https://www.adafruit.com/product/165
* https://www.adafruit.com/product/1782
* https://www.adafruit.com/product/1293
* https://www.adafruit.com/product/1120
* http://www.shop.inspeed.com/E-Vane-II-Electronic-Wind-Direction-Sensor-EV2.htm;jsessionid=28DB9A7BB5577BB5D85A400A4C2D0167.p3plqscsfapp006

kb1lqd, 2017-10-27:
Sensors received. Closing to allow integration and scalings to be tracked with more granularity after this initial selection and system design.

closed 2017-10-27 by kb1lqd