FaradayRF/Faraday-Software #267: Add I2C Sensor Firmware Support

kb1lqd, 2017-09-05:
Per #260 I2C will be needed. Either update a sharable SPI/I2C on the USC_xB port shared with the SPI SRAM or bit-bang I2C.

kb1lqc, 2017-09-08:
What about SPI to I2C converter IC's?

kb1lqd, 2017-09-08:
Damn that it's a good idea... Devil may be in the details but makes sense
to not spend time writing a bit bang solution from mostly scratch or heavy
software developer time...

I found

https://www.digikey.com/product-detail/en/nxp-usa-inc/SC18IS600IPW,112/568-3232-5-ND/1033517

Which seems like what we are looking for. I haven't found a Dev board but
maybe I should order these and a simple ic breakout?

herrold, 2017-09-08:
ummm that particular part is EOL -- link on the quoted page
   https://media.digikey.com/pdf/PCNs/NXP/201706009DN.pdf

kb1lqd, 2017-11-06:
Closing due to scope reduction, not adding I2C sensors and focusing on core telemetry and analog to then move onto other Faraday uses such as TUN network adapter and RPI integration.

closed 2017-11-06 by kb1lqd