FaradayRF/Faraday-Software #272: Check if SQLite3 WAL Mode Fixes Raspberry Pi 3 "Database Locked" Errors for Faraday-Telemetry

kb1lqc, 2017-09-10:
## Summary
Per Faraday-Software #270 testing it became apparent that Faraday-Telemetry on a Raspberry Pi Model 3 would not start. This appears to be because the database is locked. However this same code works fine on a desktop or laptop computer. Searching this issue reveals is could be that the Raspberry Pi is too slow and maybe WAL mode helps.

## Problem Explanation
[SQLite Write Ahead Logging](https://www.sqlite.org/wal.html) should drastically quicken up database operations. This could help make it work on the RPI 3 as I'd prefer no to have to move to MySQL or similar.

### Environment
#### Software 
- Faraday-Software
- Raspbian September 2017

#### Hardware
- Faraday Rev D1
- Raspberry Pi 3 Model B

# Supporting Information
Screenshot of error:
[image]

reillyeon, 2017-09-10:
Instead of creating a new SQLite3 connection for each write in `sqlInsert()` we should also try having each `telemetry_worker` connect to the database once and reuse that connection for each transaction.

kb1lqc, 2017-09-10:
Hmm I feel like there was a reason I didn't do that but that reason may no longer be valid so this is good to consider. It's driving me nuts on the Raspberry Pi. `Faraday-Telemetry` works fine on computers but not on the RPi! I'll see if this fixes it.

kb1lqc, 2017-09-14:
Noting that I should consider using a wrapper... https://rogerbinns.github.io/apsw/

reillyeon, 2017-09-14:
It's not clear to me that APSW provides any benefit over the built-in Python SQLite library.

kb1lqc, 2017-09-14:
Yeah @reillyeon that was more of a note to myself haha. Piecing through the code right now

kb1lqc, 2017-09-14:
OK I tried a bit tonight on the RPi. No success in trying to move it to a connection per thread on that. I think I should move over and implement that method on a normal computer first and then move it over to the RPi. The RPi kinda sucks as a development machine haha.....

@reillyeon I feel like the way I have it should work as most suggestions I see online say to open a new connection, execute, and immediately close.

kb1lqc, 2017-09-14:
Hmm in https://github.com/kb1lqc/Faraday-Software/commit/c4a93afbb88d1c0c4f025055b1845150f37b4896 I separated the connections but I'm still getting lock error when I run on the raspberry pi and try to query '/stations'

`2017-09-14 07:39:14,152 - Telemetry - ERROR - Sqlite3.error: database is locked`

Weird...

kb1lqc, 2017-09-14:
FIXED IT! Or at least mitigated it in https://github.com/kb1lqc/Faraday-Software/commit/32e8c303ec7cfc2b15ae3b8ad81f9eb2835a3d70 so my Raspberry Pi is running telemetry and aprs servers right now.

Actions take:
- Gave each thread a connection per @reillyeon suggestion. This alone did not work
- Enabled SQLite WAL mode every time the database is opened with `conn.execute("PRAGMA journal_mode=WAL;")`

I believe the Raspberry Pi is simply a bit too slow, at least with Raspbian running a graphical environment and `faraday-proxy` (50% CPU.....). WAL mode allows the .journal file to not be required to be removed for each access to the database. I imagine my desktop and laptop is quick enough to properly remove the .journal file before every operation but this is likely not the case on the RPI.

Will let telemetry and APRS run during the day. It's on a 10 second update rate so hopefully no one complains! I am indicating the desire to NOT be repeated over RF via APRS-IS.

https://aprs.fi/?c=raw&call=KB1LQC-1

@kb1lqd @el-iso

kb1lqc, 2017-09-15:
I turned on the APRS software this morning with my Raspberry Pi for a day long test with this modification. However it stopped a few hours later. I came home to find that my Raspberry Pi had basically halted, it needed a reboot. We will see if Faraday power draw + RPi power draw on a 2A Apple wall charger causes the Pi to periodically brownout. That's my suspicion.

It's setup for a test and what I will do in the meantime is cleanup my code and get a PR going. This is promising to say the least.

kb1lqc, 2017-09-16:
Just got to check at home the reason why it stopped transmitting to APRS-IS today:

```
2] "GET /?callsign=KB1LQC&limit=1&nodeid=2&timespan=300 HTTP/1.1" 200 -
2017-09-15 09:09:37,253 - Telemetry - ERROR - Sqlite3.error: database is locked
[2017-09-15 09:09:37,461] ERROR in app: Exception on /stations [GET]
Traceback (most recent call last):
  File "/home/kb1lqc/.local/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/kb1lqc/.local/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/kb1lqc/.local/lib/python2.7/site-packages/flask_cors/extension.py", line 161, in wrapped_function
    return cors_after_request(app.make_response(f(*args, **kwargs)))
  File "/home/kb1lqc/.local/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/kb1lqc/.local/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/kb1lqc/.local/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/kb1lqc/github/kb1lqc-faradayrf/faraday/telemetry.py", line 602, in stations
    data = queryStationsDb(parameters)
  File "/home/kb1lqc/github/kb1lqc-faradayrf/faraday/telemetry.py", line 960, in queryStationsDb
    cur = queryStationsDB.cursor()
ProgrammingError: Cannot operate on a closed database.
2017-09-15 09:09:38,952 - werkzeug - INFO - 127.0.0.1 - - [15/Sep/2017 09:09:38] "GET /stations HTTP/1.1" 500 -
```

kb1lqc, 2017-09-16:
I moved the `cur = queryStationsDB.cursor()` into the next try statement to hopefully catch this better next time. Setup for a test again. Just started. https://aprs.fi/?c=raw&limit=&call=KB1LQC-1

kb1lqc, 2017-09-16:
It stopped again this morning after a few hours of operation. See the Telemetry error below:

```
2017-09-16 04:37:41,429 - Telemetry - ERROR - Sqlite3.error: database is locked
2017-09-16 04:37:41,430 - Telemetry - ERROR - Sqlite3.error: Cannot operate on a closed database.
2017-09-16 04:37:41,430 - Telemetry - ERROR - (1505561556, 1505561856)
2017-09-16 04:37:41,431 - Telemetry - INFO - No stations have been heard in last 300 seconds
2017-09-16 04:37:41,433 - werkzeug - INFO - 127.0.0.1 - - [16/Sep/2017 04:37:41] "GET /stations HTTP/1.1" 204
```
This resulted in the following APRS error:
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/kb1lqc/github/kb1lqc-faradayrf/faraday/aprs.py", line 111, in aprs_worker
    stations = getStations()
  File "/home/kb1lqc/github/kb1lqc-faradayrf/faraday/aprs.py", line 169, in getStations
    results = r.json()
  File "/home/kb1lqc/.local/lib/python2.7/site-packages/requests/models.py", line 850, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python2.7/dist-packages/simplejson/__init__.py", line 516, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/dist-packages/simplejson/decoder.py", line 374, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python2.7/dist-packages/simplejson/decoder.py", line 404, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

kb1lqc, 2017-09-16:
OK This error has to do with `faraday-aprs` not being able to handle bad/incomplete JSON from telemetry. I was able to get the entire system working again by only starting `faraday-aprs`, leaving `faraday-telemetry` running in the same instance it did overnight.

I added `aprs.py` line 169 to a try statement, here's the end of `getStations()` in `aprs.py` with a test exception raising line to trigger the code and test the response:

```
try:
        results = r.json()
        raise ValueError("test")

    except ValueError as e:
        # JSON ValueError error, just return a blank value
        logger.error(e)
        results = ''

    # Return extracted JSON data
    return results
```

The resulting actions are good. The JSON decoder works fine and simply decodes nothing, indicating "Tracking 0 Faraday stations..." after the APRS ERROR.
``
$ faraday-aprs --callsign kb1lqc --rate 10 --start
2017-09-16 10:14:30,407 - APRS - INFO - Starting Faraday APRS-IS application
2017-09-16 10:14:30,420 - APRS - ERROR - test
2017-09-16 10:14:30,420 - APRS - INFO - Tracking 0 Faraday stations...
2017-09-16 10:14:30,420 - APRS - INFO - Connecting to APRS-IS as: KB1LQC
2017-09-16 10:14:30,500 - APRS - INFO - Connection successful!
```

Let's try this again.

kb1lqc, 2017-09-17:
Still running and looks like it survived a bad JSON!
[image]

kb1lqc, 2017-09-17:
Boom, survived locked database in telemetry applications. Proxy + Telemetry + APRS still going strong more than 24 hours after starting on the Raspberry Pi!

[image]

@kb1lqd  @el-iso  @reillyeon

kb1lqc, 2017-09-18:
Weird, everything is still technically working but APRS.fi stopped accepting telemetry:
[image]

[image]

This happened even though my software is still pumping out telemetry and positions packets right now (5:46AM):
[image]

Finally here's the aprs.fi decoded view of the latest packet:
[image]

Not sure what this is. I can't go back long enough on aprs.fi to see the exact switch over to non-working telemetry. I suspect something in my code possibly counted up/rolled over to cause an issue?

`KB1LQC-1>GPSFDY,qAS,KB1LQC:!3403.36N\11814.78W#.../000/A=000100Faraday Access Point|"1z;v8u8o!!!?!!|`

Manually using aprslib to decode the telemetry I get:
[image]

Which seems fine at first glance. I think. @reillyeon

kb1lqc, 2017-09-18:
There we go, I restarted the `faraday-aprs` program and now aprs.fi is decoding telemetry again...
`KB1LQC-1>GPSFDY,qAS,KB1LQC:!3403.36N\11814.78W#.../000/A=000100Faraday Access Point|!!;t8s8x!!!?!!|`

[image]

kb1lqc, 2017-09-18:
OK so looking at actual BASE91 telemetry, I forgot that it's every two characters to decode. So the last bad packet 
> KB1LQC-1>GPSFDY,qAS,KB1LQC:!3403.36N\11814.78W#.../000/A=000100Faraday Access Point|"1z;v8u8o!!!?!!|

BASE91 Telemetry
> "1z;v8u8o!!!?!!

Sequence = `1z` = 1545
A0 = ;v = 2451
A1 = 8u = 2177
A2 = 8o = 2171
A3 = !! = 0
A4 = !? = 30
IO = !! = 0

Yup this is wrong, there's an extra " at the beginning... what? **"**1z;v8u8o!!!?!! since all telemetry is betwee the "|" characters. What in my code eventually puts this in? It causes aprslib decoding to fail and therefore I suspect that it causes aprs.fi to fail decoding as well. It is mitigated if I use single quotes in my code but that doesn't solve the problem that there is an extra character being inserted.
[image]

Characters ( "1z;v8u8o!!!?!! ) = 15 (bad)
Characters ( !!;t8s8x!!!?!! ) = 14 (good)

It appears that it takes more than 24 hours to come up so I guess I'll just let the faraday-aprs code run again and see what happens. This might be worth a new issue ticket and tracking on there as we move along. @reillyeon @kb1lqd

kb1lqc, 2017-09-20:
Happened again roughly 30 or so hours after starting. Here's a good packet decoded by aprs.fi:

```
2017-09-19 18:47:04 PDT: KB1LQC-1>GPSFDY,qAS,KB1LQC:!3403.36N\11814.78W#.../000/A=000100Faraday Access Point|{{;v9!8)!!!>!!|
   type: location
   format: uncompressed
   srccallsign: KB1LQC-1
   dstcallsign: GPSFDY
   latitude: 34.056 °
   longitude: -118.2463333333333 °
   course: 0 °
   speed: 0 km/h
   altitude: 30.48 m
   symboltable: \
   symbolcode: #
   messaging: 0
   posresolution: 18.52 m
   posambiguity: 0
   comment: Faraday Access Point
   telemetry:
      sequence: 8280
      values: 2451, 2184, 2101, 0, 29
      bits: 00000000

2017-09-19 18:47:04 PDT: KB1LQC-1>GPSFDY,qAS,KB1LQC::KB1LQC-1 :UNIT.CNT,CNT,CNT,CNT,CNT,IO,IO,IO,IO,IO,IO,IO,IO
   type: telemetry-message
   srccallsign: KB1LQC-1
   dstcallsign: GPSFDY
   destination: KB1LQC-1
   message: UNIT.CNT,CNT,CNT,CNT,CNT,IO,IO,IO,IO,IO,IO,IO,IO

2017-09-19 18:47:04 PDT: KB1LQC-1>GPSFDY,qAS,KB1LQC::KB1LQC-1 :PARM.ADC0,ADC1,ADC3,ADC6,TEMP,IO,IO,IO,IO,IO,IO,IO,BT
   type: telemetry-message
   srccallsign: KB1LQC-1
   dstcallsign: GPSFDY
   destination: KB1LQC-1
   message: PARM.ADC0,ADC1,ADC3,ADC6,TEMP,IO,IO,IO,IO,IO,IO,IO,BT

2017-09-19 18:47:04 PDT: KB1LQC-1>GPSFDY,qAS,KB1LQC::KB1LQC-1 :EQNS.0,1,0,0,1,0,0,1,0,0,1,0,0,1,0
   type: telemetry-message
   srccallsign: KB1LQC-1
   dstcallsign: GPSFDY
   destination: KB1LQC-1
   message: EQNS.0,1,0,0,1,0,0,1,0,0,1,0,0,1,0

2017-09-19 18:47:19 PDT: KB1LQC-1>GPSFDY,qAS,KB1LQC:!3403.36N\11814.78W#.../000/A=000100Faraday Access Point|"!!;v9!8(!!!=!!|
   type: location
   format: uncompressed
   srccallsign: KB1LQC-1
   dstcallsign: GPSFDY
   latitude: 34.056 °
   longitude: -118.2463333333333 °
   course: 0 °
   speed: 0 km/h
   altitude: 30.48 m
   symboltable: \
   symbolcode: #
   messaging: 0
   posresolution: 18.52 m
   posambiguity: 0
   comment: Faraday Access Point|"!!;v9!8(!!!=!!|
```

It's important to note that the last good packet was a 10th packet so it sent the labels and equations. This is important to me being a timer/overflow issue.

kb1lqc, 2017-09-20:
Well there we go, it's a rollover in BASE91 from two characters to three:

[image]

AWE DANG I GONE MESSED UP! @kb1lqd @reillyeon @el-iso 

http://he.fi/doc/aprs-base91-comment-telemetry.txt

> 3.4 The telemetry sequence counter and telemetry channels are encoded using
the Base91 encoding, as it is presently used in Compressed APRS position
packets, the altitude extension, and the DAO extension. Two bytes are
transmitted for the sequence counter and each of the channels, giving over
13 bits of resolution (values 0 to 8280). Please note that APRS uses a
different definition of Base91 than the internet standard Base91 - see
APRS101.PDF for details.

There we go, maximum value is 8280... EXACTLY WHERE MY TELEMETRY CUTS OUT.

This is also convoluted since I believe there's also an issue with aprs.fi possibly not handling a " in the base91 string.

kb1lqc, 2017-09-20:
I should probably implement 3.6

>3.6 The telemetry sequence counter MAY wrap from 255 to 0 if memory
constraints require using a 1-byte variable for storing the counter. Please
make sure that it and all of the telemetry values never get values higher
than 8280. For example, the sequence number can be safely incremented with:

   sequence = (sequence + 1) & 0x1FFF;

This will make it wrap to 0 after 8191, which will provide plenty of range.

kb1lqc, 2017-09-21:
Implemented a for loop that counts to 8279 and resets to zero using the infinite while loop in `faraday/aprs.py` in https://github.com/FaradayRF/Faraday-Software/commit/cfdee3a9da35808b5a263ee7ffc104ce0d668d3d which should fix this problem!

kb1lqc, 2017-09-23:
Still going nearly 58.5 hours later. I had to restart networking on the Raspberry Pi as my Spectrum Internet crapped out for a few minutes today such that I needed to restart my cable modem. the Raspberry Pi wasn't pushing data to the internet until I performed a `sudo service networking restart`. The network went down as one would expect during this:
[image]

It came right back up and aprs.fi was again seeing data after a 15 minute or so loss. It's important to note that I have not restarted any of the faraday software so it's still going. Also, Faraday aprs doesn't acknowledge APRS-IS packets so it was reporting that it sent data to the server but apparently it wasn't ever leaving my local home network until I restarted the Raspberry Pi networking. This is pretty minimal effect on Faraday software so I'll disregard for now.

kb1lqd, 2017-09-28:
PR #273 merger closes this.

closed 2017-09-28 by kb1lqd