FaradayRF/Faraday-Software #274: Grafana For Faraday?

kb1lqc, 2017-09-15:
## Summary
Grafana is an open source and pretty well supported dashboard/UI for what appears to largely be datacenter type uses. However, it's touted as being useful for any time series data... AKA telemetry. This is tempting. I am thinking this might be worth looking into as opposed to NASA's OpenMCT. I am also not too sure if it can be used at all for commanding, however, this may not actually be a show-stopper depending on how easy it is to setup for viewing telemetry nicely.

[image]

Any thoughts @hdkmike @reillyeon @el-iso @kb1lqd ?

## Problem Explanation
Grafana: https://grafana.com/grafana
Grafana Demo: http://play.grafana.org/
Grafana Getting Started: http://docs.grafana.org/guides/getting_started/
Grafana Simple JSON Data Source: https://grafana.com/plugins/grafana-simple-json-datasource

### Environment
#### Software 
>Please include the branch or commit if known regarding what software was used if applicable

#### Hardware
>Let us know about what hardware you are using if applicable. This includes:
>- Faraday radio serial numbers
>- Computer operating system

# Supporting Information
>This is a handy spot to post any screenshots or file outputs in order to better support this issue. We want to keep the top of the ticket clutter-free as possible.

kb1lqc, 2017-09-15:
WHOA Found a person who uses Grafana for weather station data from CANbus sensors and an Arduino:
- Grafana: https://weather.devopstom.com/dashboard/db/weather-dashboard?orgId=1
- Project: http://tomoconnor.eu/blogish/monitoring-weather-influxdb-and-grafana-and-bunch-arduinos/
[image]

Uses:
- OpenTSDB: http://opentsdb.net/

@kb1lqd @el-iso

daniellee, 2017-09-18:
There are a couple of examples of Grafana plugins that execute commands:

- The [Influx Admin](https://grafana.com/plugins/natel-influx-admin-panel) panel lets you execute queries against InfluxDB
- The [Kubernetes App](https://grafana.com/plugins/raintank-kubernetes-app) deploys pods on the cluster.
- You can set up monitoring probes around the world with the [worldPing App](https://grafana.com/plugins/raintank-worldping-app).

So it's not the common use case for Grafana but a plugin can do anything a website can.

kb1lqc, 2017-09-19:
Great Grafana Videos:

- Installation and configuration: https://youtu.be/E-gMFv84FE8?list=PLDGkOdUX1Ujo3wHw9-z5Vo12YLqXRjzg2
- Adding data sources: https://www.youtube.com/watch?v=9ZCMVNxUf6s

kb1lqc, 2017-09-19:
Oh... We should definitely consider opentsdb: https://serverfault.com/questions/362338/what-is-the-difference-between-opentsdb-and-graphite

kb1lqc, 2017-09-19:
Looks like some good info here on creating datasources: http://docs.grafana.org/plugins/developing/datasources/

I think my first step is the fastest to proof of concept. I would love to create an opentsdb source for Faraday, I think that my be the best intermediate goal. However, the fastest path is probably to develop an aprs.fi plugin. This is actually very useful and I bet it may be popular outside of Faraday. Basically, some fun opensource work!

https://aprs.fi/page/api

kb1lqc, 2017-09-22:
While looking into this I am concerned that the startup costs with learning how a Grafana datasource plugin is constructed are relatively high. I am not an expert javascript programmer and the documentation is not terribly straightforward for the simple JSON datasource. Mostly, I am left wondering exactly what the flow of processes through the plugin are. For example it appears that the `/search` url is used to return what metrics (time series) exists which Grafana receives over a JSON response and then it proceeds to query a specific time sequence with that information. There are a lot of moving pieces.

I don't want to spend much more time on this right now and would rather move this to Buoy 2. However I want one more shot at setting up an influxdb that is sent data from telemetry. Grafana natively supports this and it may just work. This is scaleable and sets us up on a clear path to providing at least our version of a central repository for telemetry (though I would like it a bit decentralized).

https://www.influxdata.com/

kb1lqc, 2017-10-06:
With [aprs2influxdb](https://github.com/FaradayRF/aprs2influxdb) it appears that we're well on our way forward with Grafana. Closing this issue because we should start tracking this on that project issue system.

closed 2017-10-06 by kb1lqc