FaradayRF/Faraday-Software #276: Device Configuration Erroring During Runtime

kb1lqd, 2017-10-21:
## Summary
As of the last few months the device configuration software used to updated flash memory configuration of Faraday errors and does not update the configuration.

## Problem Explanation
It seems that something in the device configuration program errors, possibly due to INI file issues due to the error logging messages.

### Environment
#### Software 
Current Master as of ticket date.

#### Hardware
Faraday REV D1.

# Supporting Information

[image]

kb1lqd, 2017-10-21:
@kb1lqc I totally forgot where these files above are being copied from. I think the sample files are missing and that's why device configuration is failing to initialize and work. Can you confirm?

kb1lqd, 2017-10-21:
OK So I got proxy and device configuration running (realized I needed to run commands in the root folder only in editable mode too) and I ran into this error when trying to run simple config:

[image]

kb1lqd, 2017-10-21:
Ohhh wow I think I found something @kb1lqc 

When I init a new device configuration I see the default values:

[image]

When I update basic information I see the callsign updated with a "?" prepended...

`$ faraday-deviceconfiguration --proxycallsign "kb1lqd" --proxynodeid 1 --callsign "kb1lqd" --nodeid 1 --gpsbootoff`

[image]

kb1lqd, 2017-10-21:
OK so if I remove the `?` manually I still get the 5 arguments instead of 6 error as noted above.

kb1lqd, 2017-10-21:
OK looking into the `POST()` function to see why the incorrect number of arguments are being passed I see that something may have changed because the docstring doesn't match the number of arguments in the function definition:

``` python
def POST(self, host, local_device_callsign, local_device_id, uart_port, data):
        """
        The POST function is a python function that interacts with the Faraday RESTful API and "POSTS" (puts into) data into the transmit queue. Data provided to this function will be transmitted
        to a local Faraday device over UART (as specified by the arguments) over the intended Faraday transport layer "Service Port."

        The POST function performs all needed BASE64 encoding and JSON dictionary creation as needed by the Faraday proxy API.

        .. note:: Only a single data item is currently accepted.

        :param local_device_callsign: Callsign of the local Faraday device to direct the data to (allows multiple local units)
        :param local_device_id: Callsign ID number of the local Faraday device to direct the data to (allows multiple local units)
        :param uart_port: Intended Faraday transport layer service port to direct the supplied data to
        :param data: Data to be transmitted in string format

        :Return: Python FLASK POST status result

        :Example:

        Usually the data sent over POST will be a generated bytestring that follows the Faraday protocol stack. Below is an example of sending a command to turn ON LED #1 of the KB1LQD-1 device.

        >>> faraday_1 = faradaybasicproxyio.proxyio()
        >>> faraday_cmd = faradaycommands.faraday_commands()
        >>> command = faraday_cmd.CommandLocalGPIOLED1On()
        >>> faraday_1.POST("KB1LQD", 1, faraday_1.CMD_UART_PORT, command)
        <Response [200]>

        """
```

It looks like the `host` was added at some point and this IS NOT passed by the device configuration program. This might have been added in the push that @kb1lqc did a while ago to deconflict the multiple threads of proxy?

kb1lqd, 2017-10-21:
OK so the `host` variable missing is the IP of the proxy host in the proxyio script:

`            url = 'http://{0}:{1}/?port={2}&callsign={3}&nodeid={4}'.format(host, self.FLASK_PORT, uart_port, local_device_callsign, local_device_id)
`

I believe I just need to add this to the function call as an argument somehow in the device configuration or simpleconfig module.

kb1lqd, 2017-10-21:
OK I manually added the localhost IP to the POST commands:

[image]

The device configuration does not fail due to arguments but now waits and errors when the unit doesn't give the data back (or the data doesn't become processed correctly).

[image]

This is good! The device configuration python script has two way's of sending POST and we should standardize on one way, probably this way:

`r = requests.post('http://{0}:{1}'.format(hostname, port),
                          params={'callsign': str(local_device_callsign), 'nodeid': int(local_device_node_id), 'port': cmdPort})
`

Instead of:

`proxy.POST('127.0.0.1',str(callsign), int(nodeid), UART_PORT_APP_COMMAND,
                       faradayCmd.CommandLocalSendReadDeviceConfig())`

kb1lqc, 2017-10-21:
Yeah I think you found it. In effort to decouple local host or remote IP addresses I made this change and very likely didn't test configuration... Was back in June or July.. My bad... One more reason  for unit testing! 

---- On Sat, 21 Oct 2017 10:13:31 -0700 notifications@github.com wrote ----

OK I manually added the localhost IP to the POST commands:

The device configuration does not fail due to arguments but now waits and errors when the unit doesn't give the data back (or the data doesn't become processed correctly).

This is good! The device configuration python script has two way's of sending POST and we should standardize on one way, probably this way:

r = requests.post('http://{0}:{1}'.format(hostname, port), params={'callsign': str(local_device_callsign), 'nodeid': int(local_device_node_id), 'port': cmdPort})

Instead of:

proxy.POST('127.0.0.1',str(callsign), int(nodeid), UART_PORT_APP_COMMAND, faradayCmd.CommandLocalSendReadDeviceConfig())

kb1lqc, 2017-10-22:
Noting that when installing Faraday-Software on a clean virtualenvironment to test this problem my `six` package installed version 1.10.0 but device configuration needs 1.11.0. This is an issue with the requirements.txt file though not the original issue on this ticket.

kb1lqc, 2017-10-22:
It appears this was fixed in #277 @kb1lqd  and I updated pypi so a `pip install faraday` should install version 0.0.1017 which has the fixes for this issue in it! Please test and close.

kb1lqd, 2017-10-22:
Branch worked on my setup as intended!

kb1lqd, 2017-10-22:
@el-iso Youy should be good to go if you didn't already know from the updates.

closed 2017-10-22 by kb1lqd