FaradayRF/Faraday-Software #278: RF Station Cause APRS Application To Fail

kb1lqc, 2017-10-23:
## Summary
KB1LQC-1 Base Station was operating on a Raspeberry Pi fine for almost two weeks. Once I placed KB1LQC-2, a RF station nearby, I caused a fault to happen and the aprs application quit

## Problem Explanation
>Provide a more in-depth explanation of the problem or feature if applicable

### Environment
#### Software 
 Faraday Software V 0.0.1016

#### Hardware
>Let us know about what hardware you are using if applicable. This includes:
Rapsberry Pi 3

# Supporting Information

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/faraday/aprs.py", line 113, in aprs_worker
    stationData = getStationData(stations)
  File "/usr/local/lib/python2.7/dist-packages/faraday/aprs.py", line 213, in getStationData
    data = r.json()
  File "/home/pi/.local/lib/python2.7/site-packages/requests/models.py", line 850, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python2.7/dist-packages/simplejson/__init__.py", line 516, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/dist-packages/simplejson/decoder.py", line 374, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python2.7/dist-packages/simplejson/decoder.py", line 404, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

kb1lqc, 2017-11-06:
Again after a long time
```
2017-11-05 16:32:22,070 - APRS - INFO - Tracking 2 Faraday stations...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/faraday/aprs.py", line 113, in aprs_worker
    stationData = getStationData(stations)
  File "/usr/local/lib/python2.7/dist-packages/faraday/aprs.py", line 213, in getStationData
    data = r.json()
  File "/home/pi/.local/lib/python2.7/site-packages/requests/models.py", line 850, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python2.7/dist-packages/simplejson/__init__.py", line 516, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/dist-packages/simplejson/decoder.py", line 374, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python2.7/dist-packages/simplejson/decoder.py", line 404, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

kb1lqd, 2017-11-06:
Ok Bryce so this is looking like firmware right? Or at least software
ability to handle bad data?

Brent

reillyeon, 2017-11-06:
We should add a try/except block around JSON decoding and both ignore the error and log the malformed input for debug analysis.